pauldemarco/flutter_blue #154: What Bluetooth Specification is this work based on?

ride4sun, 2018-12-20:
I kind of new to Bluetooth and wonder what the manufacturer data is and all the other data. 
I browsed for a bluetooth spec. and I am overwhelmed how many are out there. Could you add a link to the Read me page or just add to this issue. Thanks for creating this library.

pauldemarco, 2018-12-22:
Hi @ride4sun ,

This library is based on the Core Specifications overseen by the Bluetooth Special Interest Group (SIG).
Please see here: https://www.bluetooth.com/specifications/bluetooth-core-specification

closed 2018-12-22 by pauldemarco

ride4sun, 2018-12-22:
@pauldemarco  thanks - could u add that to the readme - I am new to bluetooth and I would think it could help to understand the implementation better.