pauldemarco/flutter_blue #157: How can I identify what device I want to connect if the name data is empty

ride4sun, 2018-12-22:
If I run the example I only get data with no name info. How can I select the device I would like to connect to?
[image]

mehdimsv, 2019-04-25:
I have the same problem.
I tested the example on my two sony z3+ smart phones which have bluetooth version 4.1 but still it shows hex number instead of human readable name.
can not connect to each other, flutter_bluetooth_serial is android only and its not an option for me.
please someone  explain how can I use this package.

PsychoXIVI, 2019-05-24:
I have same problem - tested 2 android phones, raspberry Pi 3B+, HC-05 bluetooth module (Arduino) and some wireless headphones - all of the devices has no name displayed :C

enricobenedos, 2020-01-07:
Same problem, when I try to get device name the string is empty, The package provided by @bogoevskid solve this problem

ride4sun, 2020-01-07:
@enricobenedos  I think the answer is no. I basically blocked the update and use the old version. Kind of sucks.

AlexanderFilinovich, 2020-02-27:
@ride4sun Did you write that you are using the old version?
Did this solve the problem with the name? And if so, which version are you using?

jenish-rudani, 2021-02-24:
Solved for Raspberry PI 4. 

So little background on what I was trying to solve. 

My issue was to retrieve mac address of Bluetooth on Raspberry PI 4, which we used for let's say "XYZ" reason.

So, we used the flutter_blue package to set up Bluetooth and discover the raspberry PI device. All we could see was MAC Address of raspberry PI 4 but no "Local Name". Dig around a lot and found that PI 4 was not advertising BLE, so I enabled "LE Set Advertise Enable" using the following command. For more information, you can check this out [BLE Advertising](https://stackoverflow.com/a/21126744/11642764)

After enabling BLE Advertising, we could see the Local Name of Raspberry PI 4.

santigual, 2022-11-18:
I have exactly the same trouble using a Vodafone Smart N10 Android 9.0
Why I cannot read the names of the devices? using Bluetooth settings i can see it perfectly.