pauldemarco/flutter_blue #170: Build Runner fail after update from 0.4.1 to 0.4.2 with Xcode error:

koleto, 2019-01-08:
After I upgrade my project dependencies and move from flutter_blue 0.4.1 to 0.4.2 I am not able to deploy on  iOS with following errors:

```
 /usr/local/bin/pod:22:in `<main>'

Error output from CocoaPods:
↳
       /bin/bash: line 3: /Users/.../Pods/!ProtoCompiler/protoc: No such file or directory
```
 
When I open the Runner in Xcode and try to compile following error appears:

[image]

ycktw, 2019-01-08:
I've just found this issue today.

Referring to https://github.com/pauldemarco/flutter_blue/issues/40
Solution consisted of the following:

> Install cocoapods 1.2.1
> sudo gem uninstall cocoapods
> sudo gem uninstall cocoapods-core
> sudo gem uninstall cocoapods-deintegrate
> sudo gem uninstall cocoapods-downloader
> sudo gem uninstall -aIx
> sudo gem install -n /usr/local/bin cocoapods -v 1.2.1
> 
> Reinstall pods
> cd ./flutter_blue/example/ios
> rm -rf ./Podfile.lock
> export FLUTTER_FRAMEWORK_DIR=/Users/pauldemarco/flutter/bin/cache/artifacts/engine/ios
> pod install
> 
> Now, flutter run should run successfully.

and then
> brew upgrade cocoapods
> pod setup

koleto, 2019-01-08:
@ycktw  thanks a lot! This solve my problem!

closed 2019-01-08 by koleto