pauldemarco/flutter_blue #171: Can't connect with Raspberry Pi 3b+

jemutorres, 2019-01-08:
I'm trying to connect with Raspberry Pi 3b+.

The Raspberry have Raspbian Stretch Lite with Bluetooth 4.2. It's visible and I go to Settings > Bluetooth on my phone, I found the raspberry.

I downloaded the example. I've done a search from the same phone and I can't find the Raspberry.

jonasgaiswinkler, 2019-01-14:
I'm having the same issue with my Raspberry Pi Zero W

jonasgaiswinkler, 2019-01-16:
I found these instructions which work fine for me: [Bluetooth with Raspberry Pi and bleno](https://www.skyrise.tech/blog/tech/bluetooth-raspberry-pi-bleno-part-1-ibeacon/)

stuartsoft, 2019-02-25:
Ditto here to. Can we get some updates on this ticket?

stuartsoft, 2019-02-26:
@pauldemarco  If you happen to have a moment to check on this I would really appreciate it. The scanning process works fine, I'm able to see the device I want and obtain a `BluetoothDevice` object. The problem occurs when I try to actually connect to said device. Below is the log info I see when calling `flutterBlue.connect().listen()` from my Flutter app

```
D/BluetoothManager(26256): getConnectedDevices
D/BluetoothGatt(26256): connect() - device: CENSORED_MAC_ADDRESS, auto: true
D/BluetoothGatt(26256): registerApp()
D/BluetoothGatt(26256): registerApp() - UUID=CENSORED_UUID
D/BluetoothGatt(26256): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(26256): close()
D/BluetoothGatt(26256): unregisterApp() - mClientIf=10
```

and for reference, here is the snippet I'm using to initiate a connection:
```dart
_fbconnection = this
              .widget
              .flutterBlue
              .connect(this.widget.device,
                  autoConnect: true, timeout: Duration(seconds: 60))
              .listen((state) {
            setState(() {
              debugPrint("new state: " + state.toString());
              this._deviceState = state;
            });
          });
```

I've been able to confirm that I can connect my Pixel 3 XL to my Pi 3B using the following two sample projects from Google. So, that at least proves that my hardware is good.
https://github.com/androidthings/sample-bluetooth-le-gattserver
https://github.com/googlesamples/android-BluetoothLeGatt

stuartsoft, 2019-02-27:
I'm getting the impression that this library just straight up doesn't work on Android. I'm using the example scanner provided in THIS repository and it's unable to connect to my BLE Raspberry Pi. Meanwhile, Google's own BLE example works fine and connects to the same Pi running the same code. `¯\_(ツ)_/¯`

stuartsoft, 2019-02-27:
Never mind. We're all good. Turns out the connection works when you set `autoconnect: false` when attempting to connect. @pauldemarco  I guess it's just an issue when trying to connect with autoconnect on, as pointed out by this person: https://github.com/pauldemarco/flutter_blue/issues/106#issuecomment-448141259

closed 2020-04-02 by pauldemarco