pauldemarco/flutter_blue #176: Connect and discover services slow on Android

fengqiangboy, 2019-01-17:
Connect and discover services slow on Android

But fine on iOS

valebar, 2019-01-22:
I have the same problem, but only with connect service on Android. Sometimes it connects normaly but in the most cases in takes ages. I use Nokia 6 and Adafruit Feather M0 Bluefriend.

If I try to connect to Adafruit from any other app on the phone, it has no problem.

The discover service works just fine.

Does anyone have any idea what could be the problem?

valebar, 2019-01-24:
In my case, it works if I use <code>autoConnect: false</code>
Here's a snippet...
<code>deviceConnection = _flutterBlue.connect(_device, timeout: const Duration(seconds: 10), autoConnect: false)</code>

mjl0602, 2019-02-22:
If you can’t use flutterblue connect device with Android Phone. You can try low version Api of Android. These apis are deprecated but worked very well in some Android Phone. These
Phones are bad support with new API: you can call the function successful, but scan and connect will be very slow and easy connect fail. To solve this problem, change file:
 android/src/main/java/com/pauldemarco/flutterblue/

see new file on my fork: https://github.com/mjl0602/flutter_blue.git

This change didn't solve this problem completely. To solve this problem, must add new args to control the api version. but not use Build.VERSION.SDK_INT. It's works bad.

简单的说，就是很多手机系统到了新版本，但是对新版本的硬件API支持的很差，强制换成老版本的用法就好了。希望作者加一个字段来控制具体用什么版本的API来搜索。

andretietz, 2019-04-26:
I think I found the problem, but don't know how to fix it. I also have this problem that often connecting is very slowly.

When debugging the connection I figured out, that if I set a breakpoint at [line 133](https://github.com/pauldemarco/flutter_blue/blob/master/lib/src/flutter_blue.dart#L133) the connection is established almost immediately. 
Alternatively I added a `sleep(const Duration(milliseconds: 1000));` before the subscription. This is solving the issue even though it needs at least one second.(Note: this is for figuring out what's wrong and shouldn't be part of the implementation)

Can anyone confirm that?

If so, the problem is listening to the devices state-changes "too early", meaning that the connection is not established, but we're listening to the changes. If so we're not getting the first state update and it 'feels' like we're not connected even though we might be.

I

Drkstr, 2020-01-23:
Based on [this](https://stackoverflow.com/questions/22214254/android-ble-connect-slowly) StackOverflow answer, it looks like setting the autoConnect parameter int he connect method to false makes the Android connect as fast as iOS.