pauldemarco/flutter_blue #177: what's the format of writeCharacteristic'

100cm, 2019-01-22:


lemonbuzz, 2019-01-22:
I think you want something like this in the array
[0x05,0x66,0xA0,0x33,0x0A]

100cm, 2019-01-23:
@lemonbuzz thanks a lot ! dude! but it must in hexadecimal?
my workmate write a java function for me 

```
    public static byte[] hex2byte(String hex) {
        String digital = "0123456789ABCDEF";
        char[] hex2char = hex.toCharArray();
        byte[] bytes = new byte[hex.length() / 2];
        int temp;
        for (int i = 0; i < bytes.length; i++) {
            temp = digital.indexOf(hex2char[2 * i]) * 16;
            temp += digital.indexOf(hex2char[2 * i + 1]);
            bytes[i] = (byte) (temp & 0xff);
        }
        return bytes;
    }
```

but the result is `[5, 102, -96, 51, 10]`

and  `[0x05,0x66,0xA0,0x33,0x0A]` equals   `[5, 102, 160, 51, 10]`

i don't why the third number is different.

lemonbuzz, 2019-01-23:
Take a look at this
https://api.dartlang.org/stable/2.1.0/dart-convert/Encoding-class.html

encode method returns the buffer you can pass to writeCharacteristics

Pretty useful with ascii, i use it to send commands to my esp32 app :)

d3rf, 2019-02-04:
My ESP is expecting for String, so how create a Method to write as String ?

jpcEngineer, 2019-02-04:
That list of int's are Hex,
[image]

100cm, 2019-02-18:
finally i solved this problem. close the issue.

closed 2019-02-18 by 100cm

kamilk91, 2019-10-18:
@100cm  how? i have the same problem now.