pauldemarco/flutter_blue #178: Can't connect to devices with example app

kattzor, 2019-01-23:
I'm new to Flutter and and wanted to test out some of the BLE functionality that Flutter Blue offers. I've got the example application up and running and it finds plenty of devices around me. The problem is that I'm not able to connect to any of them. I'm currently using a second phone to emulate a BLE Peripheral, all other software such as nRF Connect for Mobile is able to connect to the peripheral and read the data it provides. The Flutter Blue example application is just stuck on disconnected when attempting to connect to the device.

Here is a typical console output when attempting to connect:

D/BluetoothGatt( 6359): connect() - device: 73:B9:9F:25:67:DC, auto: true
D/BluetoothGatt( 6359): registerApp()
D/BluetoothGatt( 6359): registerApp() - UUID=44fdc1ff-3804-4ae2-ab52-3f3b6eb31e17
D/BluetoothGatt( 6359): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt( 6359): onAutoConnectionStatusCb()status =10
D/BluetoothGatt( 6359): close()

I have managed to connect to the emulated peripheral a few times by first connecting via a different application and then "connecting" with the Flutter application. This is the output:

D/BluetoothGatt( 6359): connect() - device: 73:B9:9F:25:67:DC, auto: true
D/BluetoothGatt( 6359): registerApp()
D/BluetoothGatt( 6359): registerApp() - UUID=201eaf3d-ab8a-402e-9cbf-232527fa53a8
D/BluetoothGatt( 6359): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt( 6359): onAutoConnectionStatusCb()status =10
D/BluetoothGatt( 6359): onClientConnectionState() - status=0 clientIf=8 device=73:B9:9F:25:67:DC
D/FlutterBluePlugin( 6359): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt( 6359): discoverServices() - device: 73:B9:9F:25:67:DC
I/flutter ( 6359): connected!
D/BluetoothGatt( 6359): onSearchComplete() = Device=73:B9:9F:25:67:DC Status=0
D/FlutterBluePlugin( 6359): [onServicesDiscovered] count: 3 status: 0

I have read about connections not closing properly after exiting other applications and I've made sure to toggle both Bluetooth and Location services on/off as well as uninstalling other applications that might interfere with the Flutter application. I've tried allowing a longer timeout duration on connecting to make sure that the application doesn't close the connection before being able to actually connect, apart from that I'm a bit lost and can't quite put my finger on what might be wrong.

kattzor, 2019-01-26:
Just wanted to add that this seems to have been a issue related to the emulated BLE peripheral. I've since been able to make it work with both an AT-09 module and a ESP32-board without problems!

closed 2019-01-26 by kattzor

a01271588, 2019-03-12:
Hi kattzor im currently trying to make this library work with an esp32 board, i can make the connection between my android device and the board but i cant send characteristics using the "writeCharacteristic"method.
i get this error:
`E/flutter (11069): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: NoSuchMethodError: Class '_ControllerSubscription<BluetoothDeviceState>' has no instance method 'writeCharacteristic'.
E/flutter (11069): Receiver: Instance of '_ControllerSubscription<BluetoothDeviceState>'
E/flutter (11069): Tried calling: writeCharacteristic("a", Instance(length:2) of '_GrowableList')
E/flutter (11069): #0      Object.noSuchMethod (dart:core/runtime/libobject_patch.dart:50:5)
E/flutter (11069): #1      _MyHomePageState.build.<anonymous closure> (package:bt_test/main.dart:65:24)
E/flutter (11069): <asynchronous suspension>`

and this is the example code i use to try it [https://github.com/abc873693/Flutter-bluetooth/blob/master/lib/main.dart](url), any ideas? I'm actually new to flutter. Thank You

neerajyadv, 2019-03-30:
This is not working correctly. We need to connect with other device manually by bluetooth and then this service will connect. Any news on this?