pauldemarco/flutter_blue #179: Characteristic Mapping not accurate with BLE device

rubberbird, 2019-01-30:
It could be me, or it could be the plugin, but the response after subscribing to a characteristic isn't aligning with the related GATTs profile.

Details in this issue.

https://stackoverflow.com/questions/54427537/understanding-ble-characteristic-values-for-cycle-power-measurement-0x2a63

okachobi, 2019-01-30:
I don't know the answer, but you realize that 52 is a single byte.  It is 8 bits, not 16.  I read your post and went to the spec and it said that the first 16 bits are the flags.   In your first image, the first 16 bits are 52,24 or 0x3418.  In binary that is 11010000011000.  I don't know if that makes any more sense to you in the context of the cycle power management, but its not 0x0034 for the flags.  Its 0x3418 (assuming big endian byte order for words)

okachobi, 2019-01-30:
I think what you are missing is the column in the spec that says "format".  The flags take 16 bits and tell you which other values are present in the output.  The first 16 bit value is mandatory- always there regardless of the flags. The 2nd 16 bits (2 bytes) is also mandatory.  The bits tell you which other fields are present and in what order, and if you read the spec carefully you will see that some of them at 16 bits (2 byte values) and others are 24 bits (3 byte values) and even others are 12 bits (2 of them are packed into 3 byte values).  What you see in the screenshots are single bytes - 8 bits per number output.  If you map the bits in the first 2, and then pay attention to the description of the field sizes (where it says C1/C2 etc...look in the left cell for the description), I suspect it will make more sense. uint16 = 2 byte values.  uint24 = 3 byte values.  uint12 = 1.5 bytes. etc...

rubberbird, 2019-01-30:
Ah ok, I see what you are getting at. To be honest this bit stuff is doing my head in. I will do what you said and try to map it out like that. If the value, for example is Wheel Revolution Data which is uint32 32 bits, then it should be 3 of the values in the list? However the document says " C1:When present, these fields are always present as a pair." which is confusing. Also, the exponent is suppose to be a decimal which is the value I actually need.

okachobi, 2019-01-30:
Close, I think you just mistyped- uint32 is 32 bits, or 4 values (4x8=32).  A couple of them are not multiples of 8 though- the uint12 values for example.  2 of these values add up to 24 bits, or 3 8 bit values.

To your point, the exponents in binary are confusing to me too.  I don't read these specs much, so I don't know exactly what it means, but presumably these values are in decimal units that are often < 0.0 and the negative exponent is what you use to retrieve the ACTUAL decimal value once you decode the integer value.  But don't take my word at that, and the binary exponents I would assume are bitshifts, whereas the decimal exponents are calculated with the math function exp().  So I'd assuming if you see Exponent: binary -11 that it would mean the value you retrieve has to be shifted 11 bits to the right.  ex: if the value is 10, then you'd need to do ((double) 10) >> 11 to get the value. It might also be float instead of double- the specs on bluetooth should provide guidance. But again, I'm guessing- maybe someone else who is more familiar with the bluetooth specs can say what this means.

rubberbird, 2019-01-30:
I am starting to understand, but still far away from being able to map those fields. I think I need to do some research. Thanks for the taking the time to respond, I appreciate your help.