pauldemarco/flutter_blue #183: Connection speed is slow

roamboy, 2019-02-06:
Connection speed is slow， Is there a problem with the settings?

ened, 2019-02-08:
@roamboy What is your benchmark? Does it work faster with other BLE libraries? Which platform, phone, device.. ?

roamboy, 2019-02-13:
I was test connect to nrf51822, some time needs about 10s, some time about 30s, some time about 50s, some time unable to connect. It only needs about 3 ~5s, when ues nRF Toolbox.

roamboy, 2019-02-13:
@ened I was test connect to nrf51822, some time needs about 10s, some time about 30s, some time about 50s, some time unable to connect. It only needs about 3 ~5s, when ues nRF Toolbox.

mjl0602, 2019-02-22:
you can see my comment.
https://github.com/pauldemarco/flutter_blue/issues/106#issuecomment-448141259