pauldemarco/flutter_blue #185: device.readDescriptor() only returns first byte

raphael-bmec-co, 2019-02-07:
Hi

I am trying to set up bi-directional communication with a Bluno Nano.

Sending an array of bytes to the Bluno is working 100%.

Trying to receive data on that service & characteristic is giving me some issues:

1. I cannot use the `device.setNotifyValue(characteristic, true);` as the characteristic/service  is unrecognised and throws and exception. 

> Service UUID :  0000dfb0-0000-1000-8000-00805f9b34fb
> Characteristic UUID : 0000dfb1-0000-1000-8000-00805f9b34fb

2. When I attempt to use `device.readDescriptor(d);` I only get the first byte instead of the set of bytes.

If you have any guidance that would be awesome.

vikramkapoor, 2019-09-07:
@raphael-bmec-co  I am running into this issue with an iTag button only on Android. It works fine on iOS. How did you hardcode the cccDescriptor ?

raphael-bmec-co, 2019-09-08:
@vikramkapoor 

Code extract below:

const String BLUNO_SERVICE = "0000dfb0-0000-1000-8000-00805f9b34fb";
const String BLUNO_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb";
const String BLUNO_CCCD_DESCRIPTOR = "00002902-0000-1000-8000-00805f9b34fb";

BluetoothCharacteristic characteristic = BluetoothCharacteristic(
    uuid: Guid(BLUNO_CHARACTERISTIC),
    serviceUuid: Guid(BLUNO_SERVICE),
    descriptors: [
      BluetoothDescriptor(
        uuid: Guid(BLUNO_CCCD_DESCRIPTOR),
        serviceUuid: Guid(BLUNO_SERVICE),
        characteristicUuid: Guid(BLUNO_CHARACTERISTIC),
      ),
    ],
    properties: CharacteristicProperties(
      read: true,
      writeWithoutResponse: true,
      write: true,
      notify: true,
    ),
  );

raphael-bmec-co, 2019-09-26:
Just a ping here. Can we look to implement a fix on this?

aytunch, 2019-11-04:
I am having the same issue with MiBand3. I would be really glad if this fix becomes permanent in flutter_blue. I am getting crashes all over the place:( @raphael-bmec-co do you have a separate branch for the fix?

vikramkapoor, 2019-11-04:
All I did to get around this issue was to change an exception to a print statement so it doesn't crash.

raphael-bmec-co, 2019-11-05:
This is the "correct" solution"All I did to get around this issue was to
change an exception to a print statement so it doesn't crash."
--
Raphael Smith | Managing Director
[image: | BMEC Technologies |]
Ethical Elegant Engineering
+2779 496 5971
bmec.co

-- 
*Disclaimer: *The information in this e-mail is confidential. It is 
intended solely for the addressee. Access to this e-mail by anyone else is 
unauthorised. If you are not the intended recipient, any disclosure, 
copying, distribution or any action taken or omitted in reliance on it, is 
prohibited and may be unlawful. Whilst all reasonable steps are taken to 
ensure the accuracy and integrity of information and data transmitted 
electronically and to preserve the confidentiality thereof, no liability or 
responsibility whatsoever is accepted if information or data is, for 
whatever reason, corrupted or does not reach its intended destination. If 
you have received this email in error, please notify the sender immediately 
by e-mail, fax or telephone and thereafter return and/or destroy the 
original hereof.

brunotacca, 2020-05-19:
Same issue here. Any fix?

> D/FlutterBluePlugin(21510): [onServicesDiscovered] count: 3 status: 0
> I/flutter (21510): Service: 00001801-0000-1000-8000-00805f9b34fb
> I/flutter (21510): --------------------
> I/flutter (21510): Service: 00001800-0000-1000-8000-00805f9b34fb
> I/flutter (21510): --------------------
> I/flutter (21510): Service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
> I/flutter (21510): --------------------
> I/flutter (21510): > Characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
> I/flutter (21510): TRANSMITTER!
> I/flutter (21510): > > Properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: true, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
> I/flutter (21510): > --------------------
> I/flutter (21510): > --------------------
> I/flutter (21510): > Characteristic: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
> I/flutter (21510): RECEIVER!
> D/BluetoothGatt(21510): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
> I/flutter (21510): > > Properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: true, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
> I/flutter (21510): > --------------------
> I/flutter (21510): > > Descriptor: BluetoothDescriptor{uuid: 00002902-0000-1000-8000-00805f9b34fb, deviceId: E3:84:01:B2:21:0A, serviceUuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: []}
> 

Then... when I try to setNotify on the trasmitter characteristic...
`characteristic.setNotifyValue(true)`

> D/FlutterBluePlugin(21510): [onDescriptorWrite] uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
> I/flutter (21510): PlatformException(set_notification_error, could not locate CCCD descriptor for characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, null)
> 

And even, when I try to write to the receiver, I get...
`characteristic.write(utf8.encode("hello"))`

> E/flutter (21510): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)

This isn't related to BLUNO tho, I am trying to work with a custom Arduino BLE with UART.
The android app "nRF UART v2.0" works, so I am guessing I am missing something with flutter blue.

Thanks in advance.

raphael-bmec-co, 2020-05-19:
If you are using the Arduino BLE library you need to set the 2902 descriptor for each characteristic as is done in the uart example provided with the library.

`pCharacteristic->addDescriptor(new BLE2902());`

brunotacca, 2020-05-19:
> If you are using the Arduino BLE library you need to set the 2902 descriptor for each characteristic as is done in the uart example provided with the library.
> 
> `pCharacteristic->addDescriptor(new BLE2902());`

I am using the STBLE library example (TinyZero + TinyBLE).
I was curious about the descriptor you mentioned because I found the descriptor mentioned at the receiver address... 
So, after some research about descriptors and after digging through the STBLE example code, I also found the method that writes the data, is labeled TX, but it uses the RXHandler

```
uint8_t Write_UART_TX(char* TXdata, uint8_t datasize)
{
  tBleStatus ret;

  ret = aci_gatt_update_char_value(UARTServHandle, UARTRXCharHandle, 0, datasize, (uint8_t *)TXdata);

  if (ret != BLE_STATUS_SUCCESS) {
    PRINTF("Error while updating UART characteristic.\n") ;
    return BLE_STATUS_ERROR ;
  }
  return BLE_STATUS_SUCCESS;

}
```

I also couldn't found where the STBLE example sets 2902 the descriptor for the characteristic, I guess it's predefined somewhere else, maybe at the STBlueNRG library... (https://github.com/TinyCircuits/TinyCircuits-TinyShield-BLE-ASD2116/tree/master/examples/STBLE/examples/UARTPassThrough)

So, in the flutter app after swapping the characteristics order, and using the one with the 2902 descriptor to receive the data, it worked. I don't know if the example is wrong, or, it's supposed to be swapped by the UART standards...

Anyways, thank you so much.

teekoksiang, 2020-06-15:
> @vikramkapoor
>
> Code extract below:
>
> const String BLUNO_SERVICE = "0000dfb0-0000-1000-8000-00805f9b34fb";
> const String BLUNO_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb";
> const String [...]

@raphael-bmec-co The class BluetoothCharacteristic doesn't have a default constructor now. How did you create the CCCD descriptor for the BluetoothCharacteristic now or you just log the error statement?

raphael-bmec-co, 2020-06-15:
Hi

We've moved onto ESP32 so we no longer have this issue. To be honest, the server should provide that descriptor, so I'm not sure it's a reasonable expectation for flutter_ble to support this use case.

MrCsabaToth, 2020-09-03:
> @vikramkapoor
> 
> Code extract below:
> 
> const String BLUNO_SERVICE = "0000dfb0-0000-1000-8000-00805f9b34fb";
> const String BLUNO_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb";
> const String BLUNO_CCCD_DESCRIPTOR = "00002902-0000-1000-8000-00805f9b34fb";

Is there a methodical way to determine this CCCD? I get the same error for a Precor Spinning Power Chrono console. I don't have any documentation on CCCD descriptors, let alone service GUIDs and characteristics GUIDs...

raphael-bmec-co, 2020-09-03:
@MrCsabaToth I'm afraid the correct answer to your question is to readup on BLE GATT and once you have a clear understanding of that you will understand the need for the server to provide these descriptors. If the device and/or library you are using is unable to, I would recommend finding another device/library.

vikramkapoor, 2020-09-03:
It's been a while since I have looked at this code but in my use case just changing the exception to a debug log was a sufficient fix.

raphael-bmec-co, 2020-09-08:
To celebrate we are exhibiting some of our client's amazing projects...

It is our 5th anniversary!
To celebrate we are exhibiting some of our client's amazing projects on instagram (https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=31d1e783bf&e=1221793728) .
Click on any cover pic below to see the full post
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=664ca505de&e=1221793728
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=ec6ae3acbf&e=1221793728
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=5b4e63810c&e=1221793728
mailto:info@bmec.co

https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=22937a8303&e=1221793728

Suction dissector - a blunt dissector with integrated suction...

https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=43dd7efc3d&e=1221793728

Any guesses?

https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=dd942ee62c&e=1221793728

SmartSpreader - a electromechanical GPS-enabled precision farming system that...

https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=c19602a4db&e=1221793728

SmartSocket - conceptualised at the height of load shedding this wall socket...

https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=68475f872d&e=1221793728
Oculosubarachnoid shunt - a novel surgical implant for the treatment of...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=ec19ebbe3c&e=1221793728
Optic nerve bench tester - a test rig for...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=87977057b5&e=1221793728
Water management device - an IoT node for tank-level sensing, flow-rate monitoring and...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=15fbe2f968&e=1221793728
Wind correction system - an electronic system that corrects for wind offset in...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=9da7704468&e=1221793728
Oculosubarachnoid shunt deployment system - a medical device for the deployment...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=5b1b8f36f4&e=1221793728
BikeRV - an electric assist recumbent trike you can camp in!
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=be73df49f7&e=1221793728
Orbital retractor - a novel surgical instrument that enables...
https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=571de5629f&e=1221793728
Industrial Ozone Sanitation System - an ozone-based sanitation system for food processing...
Contact us (mailto:info@bmec.co?subject=Anniversary%20email%20follow-up)

============================================================
** Instagram (https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=ece9308432&e=1221793728)
** Website (https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=e156b94b38&e=1221793728)
** LinkedIn (https://bmec.us17.list-manage.com/track/click?u=1bc589e22bf937b7a9eae02b9&id=404e0edb5d&e=1221793728)
** Email (mailto:info@bmec.co)
Copyright © 2020 BMEC Technologies Pty Ltd, All rights reserved.

This email was sent to reply@reply.github.com (mailto:reply@reply.github.com)
why did I get this? (https://bmec.us17.list-manage.com/about?u=1bc589e22bf937b7a9eae02b9&id=08ebaf1385&e=1221793728&c=2853997c09)     unsubscribe from this list (https://bmec.us17.list-manage.com/unsubscribe?u=1bc589e22bf937b7a9eae02b9&id=08ebaf1385&e=1221793728&c=2853997c09)     update subscription preferences (https://bmec.us17.list-manage.com/profile?u=1bc589e22bf937b7a9eae02b9&id=08ebaf1385&e=1221793728)
BMEC Technologies . 6 Aliwal Rd . Kenilworth . Cape Town, Western Cape 7708 . South Africa

Email Marketing Powered by Mailchimp
http://www.mailchimp.com/email-referral/?utm_source=freemium_newsletter&utm_medium=email&utm_campaign=referral_marketing&aid=1bc589e22bf937b7a9eae02b9&afl=1

MrCsabaToth, 2021-06-13:
> Hi
> 
> We've moved onto ESP32 so we no longer have this issue. To be honest, the server should provide that descriptor, so I'm not sure it's a reasonable expectation for flutter_ble to support this use case.

In any way flutter_blue shouldn't poop itself even if there are bad actors around. A bad BLE implementation basically Denial of Services the plugin and it crashes? Not good.

MuFengYi, 2021-11-05:
List<int> value = await c.read();
 print("value=======" + value.toString());
c.value.listen((value) {  
});
iOS can get the data all the time, but Android cannot get this data and data

MrCsabaToth, 2021-11-05:
> List value = await c.read(); print("value=======" + value.toString()); c.value.listen((value) { }); iOS can get the data all the time, but Android cannot get this data and data

@MuFengYi  Just a thought: have you tried to just listen? I never combine read and then listen. I read (single values) or listen (to a stream) but I don't mix those.

MuFengYi, 2021-11-08:
I tried to modify it, but the value cannot be read. There is a message notification but no value. My Bluetooth read-write characteristic has no notifiy and 2902 description. I now use flutter_reactive_ble for Android and use flutter_blue for iOS.

MrCsabaToth, 2021-11-08:
> flutter_reactive_ble

Thanks for pointing me to flutter_reactive_ble. What's the reason you aren't using that for iOS? Does it have too many problems with iOS (I see quite a few issues). It's a little concerning that flutter_ble_lib which is Polidea's own plugin (flutter_reactive_ble is also based on Polidea's RxAndroidBle. I'm running my own fork of flutter_blue (https://github.com/CsabaConsulting/flutter_blue/) for a good while now and I'm keeping an eye out for alternatives. My app has 40 dependencies, I had to fork more than half a dozen of those (including bluetooth_enable as well) and I'd rather like to focus on my own app than dealing with deprecated Android API calls, v1 Flutter embeddings and crafting PRs for plugins.

MuFengYi, 2021-11-17:
Because I need the value of manufacturerData, I need to change the source code to use this value on flutter_reactive_ble on iOS.