pauldemarco/flutter_blue #189: uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue]

Sm03leBr00t, 2019-02-13:
First things frst great plugin, but it doesnt work correctly by default :/ Pls help me I am a total newcommer and want to open locks we build in the company I do a practica at. I always get the following error:
`
* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue] C:\Users\Pi-Gast\Desktop\piBike-Lock_Alex\pibikeapp\build\flutter_blue\intermediates\manifests\full\debug\AndroidManifest.xml as the library might be using APIs not available in 16
  	Suggestion: use a compatible library with a minSdk of at most 16,
  		or increase this project's minSdk version to at least 19,
  		or use tools:overrideLibrary="com.pauldemarco.flutterblue" to force usage (may lead to runtime failures)
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
* Get more help at https://help.gradle.org
BUILD FAILED in 3s
Gradle task assembleDebug failed with exit code 1
Exited (sigterm)`

GauravPatni, 2019-02-14:
Change minSdkVersion to 19  to your Android -src -builf.gradle file

Also check if scanning works properly on Android 4.4.4 devices as I am facing issue for this https://github.com/pauldemarco/flutter_blue/issues/186

Sm03leBr00t, 2019-02-14:
When I change it I get the following error:
```
Launching lib\main.dart on SM G920F in debug mode...
C:\Users\Pi-Gast\.gradle\caches\transforms-1\files-1.1\core-1.0.1.aar\e3de24e000f155198a23b39d789f7dc8\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/fontVariationSettings not found.
C:\Users\Pi-Gast\.gradle\caches\transforms-1\files-1.1\core-1.0.1.aar\e3de24e000f155198a23b39d789f7dc8\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/ttcIndex not found.
C:\Users\Pi-Gast\Desktop\piBike-Lock_Alex\pibikeapp\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:20: error: resource android:attr/fontVariationSettings not found.
C:\Users\Pi-Gast\Desktop\piBike-Lock_Alex\pibikeapp\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:20: error: resource android:attr/ttcIndex not found.
error: failed linking references.
Failed to execute aapt
com.android.ide.common.process.ProcessException: Failed to execute aapt
	at com.android.builder.core.AndroidBuilder.processResources(AndroidBuilder.java:796)
	at com.android.build.gradle.tasks.ProcessAndroidResources.invokeAaptForSplit(ProcessAndroidResources.java:551)
	at com.android.build.gradle.tasks.ProcessAndroidResources.doFullTaskAction(ProcessAndroidResources.java:285)
	at com.android.build.gradle.internal.tasks.IncrementalTask.taskAction(IncrementalTask.java:109)
	at sun.reflect.GeneratedMethodAccessor668.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
	at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$IncrementalTaskAction.doExecute(DefaultTaskClassInfoStore.java:173)
	at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.execute(DefaultTaskClassInfoStore.java:134)
	at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.execute(DefaultTaskClassInfoStore.java:121)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:122)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:105)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:99)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.execute(DefaultTaskExecutionPlan.java:625)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.executeWithTask(DefaultTaskExecutionPlan.java:580)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.run(DefaultTaskPlanExecutor.java:99)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:63)
	at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:46)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:55)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details
	at com.google.common.util.concurrent.AbstractFuture.getDoneValue(AbstractFuture.java:503)
	at com.google.common.util.concurrent.AbstractFuture.get(AbstractFuture.java:482)
	at com.google.common.util.concurrent.AbstractFuture$TrustedFuture.get(AbstractFuture.java:79)
	at com.android.builder.core.AndroidBuilder.processResources(AndroidBuilder.java:794)
	... 47 more
Caused by: java.util.concurrent.ExecutionException: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details
	at com.google.common.util.concurrent.AbstractFuture.getDoneValue(AbstractFuture.java:503)
	at com.google.common.util.concurrent.AbstractFuture.get(AbstractFuture.java:462)
	at com.google.common.util.concurrent.AbstractFuture$TrustedFuture.get(AbstractFuture.java:79)
	at com.android.builder.internal.aapt.v2.QueueableAapt2.lambda$makeValidatedPackage$1(QueueableAapt2.java:179)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	... 1 more
Caused by: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details
	at com.android.builder.png.AaptProcess$NotifierProcessOutput.handleOutput(AaptProcess.java:463)
	at com.android.builder.png.AaptProcess$NotifierProcessOutput.err(AaptProcess.java:415)
	at com.android.builder.png.AaptProcess$ProcessOutputFacade.err(AaptProcess.java:332)
	at com.android.utils.GrabProcessOutput$1.run(GrabProcessOutput.java:104)
FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':app:processDebugResources'.
> Failed to execute aapt
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
* Get more help at https://help.gradle.org
BUILD FAILED in 9s
Gradle task assembleDebug failed with exit code 1
Exited (sigterm)
```

Sm03leBr00t, 2019-02-14:
Ok you dont only have to change the `minSdkVersion` to `19` no you also have to change the `compiled` and `targetSdkversion` to `28` then it works as usual 👍

closed 2019-02-14 by Sm03leBr00t

kartikbathla, 2019-07-22:
can it be done the same way on flutter as android? I tried editing the xml as one would do in android studio and it didnt resolve the issue. Can you please share your xml code

Sm03leBr00t, 2020-03-31:
@kartikbathla Ohh man I am sooo sorry for not answering you 😞 I hope your issue is resolved now! Especially after the new pull with the Permission fixes! If not feel free to contact me

brunoleone, 2020-04-10:
olá, você permanece com o problema? Se sim, tente adcionar o seguinte codigo ao arquivo _android/gradle.properties_: 

**android.useAndroidX=true
android.enableJetifier=true**

gujianxing, 2020-05-18:
> minSdkVersion

It works for me. open /android/app/build.gradle than update minSdkVersion to 19.

insinfo, 2022-01-04:
I'm seeing this problem too

```console
Launching lib\main.dart on Android SDK built for x86 in debug mode...
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
C:\MyDartProjects\shader_compilation_demo\android\app\src\debug\AndroidManifest.xml Error:
	uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:webview_flutter_android] C:\MyDartProjects\shader_compilation_demo\build\webview_flutter_android\intermediates\library_manifest\debug\AndroidManifest.xml as the library might be using APIs not available in 16
	Suggestion: use a compatible library with a minSdk of at most 16,
		or increase this project's minSdk version to at least 19,
		or use tools:overrideLibrary="io.flutter.plugins.webviewflutter" to force usage (may lead to runtime failures)

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugMainManifest'.
> Manifest merger failed : uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:webview_flutter_android] C:\MyDartProjects\shader_compilation_demo\build\webview_flutter_android\intermediates\library_manifest\debug\AndroidManifest.xml as the library might be using APIs not available in 16
  	Suggestion: use a compatible library with a minSdk of at most 16,
  		or increase this project's minSdk version to at least 19,
  		or use tools:overrideLibrary="io.flutter.plugins.webviewflutter" to force usage (may lead to runtime failures)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 16s
The plugin webview_flutter_android requires a higher Android SDK version.
Fix this issue by adding the following to the file C:\MyDartProjects\shader_compilation_demo\android\app\build.gradle:
android {
  defaultConfig {
    minSdkVersion 19
  }
}
Note that your app won't be available to users running Android SDKs below 19.
Alternatively, try to find a version of this plugin that supports these lower versions of the Android SDK.
Exception: Gradle task assembleDebug failed with exit code 1
Exited (sigterm)
```

mohammedek, 2022-07-09:
No fix for flutter 3

kyiroywops, 2024-09-03:
still cant fix it ;(