pauldemarco/flutter_blue #196: need reconnect

chen28683, 2019-02-25:
I think this code have some problem:
// If device was connected to previously but is now disconnected, attempt a reconnect
                if(mGattServers.containsKey(deviceId) && !isConnected) {
                    if(!mGattServers.get(deviceId).connect()){
                        result.error("reconnect_error", "error when reconnecting to device", null);
                        return;
                    }
                }

if the connect() return false, we need call the connectGatt and replace old gatt object, not throw an exception

GordonHuangYong, 2019-03-26:
i do think so

bretie, 2019-06-18:
I'm running into this issue as well. Did you fix this?

xuzhengquan, 2019-07-01:
> I'm running into this issue as well. Did you fix this?

Hey, did you fix this? i have same problem.

zubairehman, 2019-07-11:
i am getting the same exception

RafaRuiz, 2020-03-22:
@chen28683 could you send a PR to fix this?

RafaRuiz, 2020-03-22:
For whoever hits this issue, I was hitting this when I was connecting to a device, then the Bluetooth gets disconnected and connected back again.
Calling `device.connect()` wouldn't work, it was throwing this error. 
My fix was calling `device.disconnect().then((_) { device.connect();});`