pauldemarco/flutter_blue #199: Version 0.5.0 causing gradle error: Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'

cujslive, 2019-02-28:
Using `flutter_blue` version `0.5.0` causes gradle error when runing app in both debug and release mode.

## Complete Log
```
Running "flutter packages get" in diabeez-patient...                1.0s
Initializing gradle...                                              0.5s
Resolving dependencies...                                           2.3s
Launching lib/main.dart on Redmi Note 4 in release mode...
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
Note: /home/chandu/Flutter/.pub-cache/hosted/pub.dartlang.org/firebase_messaging-1.0.5/android/src/main/java/io/flutter/plugins/firebasemessaging/FlutterFirebaseInstanceIDService.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.                    
Note: /home/chandu/Flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.                    
Note: /home/chandu/Developer/Flutter/diabeez-patient/build/flutter_blue/generated/source/proto/release/javalite/com/pauldemarco/flutterblue/Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.                      
Note: /home/chandu/Flutter/.pub-cache/hosted/pub.dartlang.org/video_player-0.6.5/android/src/main/java/io/flutter/plugins/videoplayer/VideoPlayerPlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.                    
Request to incrementing alive workforce from 0. Current workforce (dead or alive) 0
thread-pool size=4                                                      
ProGuard, version 6.0.3                                                 
Reading input...                                                        
Reading program jar [/home/chandu/Flutter/bin/cache/artifacts/engine/android-arm64-release/flutter.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/exoplayer-hls-2.8.0.aar/75f80f88baf3ad4f929d268feebb7beb/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/exoplayer-dash-2.8.0.aar/6a7497aa5bbd25e815b26a09c06904d8/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/exoplayer-smoothstreaming-2.8.0.aar/76dc3e4cd7bff9394ef4669bb258abef/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/exoplayer-core-2.8.0.aar/eee9ef1aeb11c325e5bc460290186b6b/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/launcher-1.2.0.aar/47c0c20a44ad51bf22bce3d2599b3814/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/firebase-messaging-17.1.0.aar/9b2443a62ca3306148deaeb317f1ca1f/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/core-1.0.1.aar/21c3dfbe63055d5ba4e86274d9803e51/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protobuf-lite/3.0.1/59b5b9c6e1a3054696d23492f888c1f8b583f5fc/protobuf-lite-3.0.1.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/design-27.1.1.aar/c2953fdc269b2b66403609f093b96854/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/firebase-iid-16.2.0.aar/9c0a133635dfa28efb10230080689dbe/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/firebase-common-16.0.0.aar/625be630717fbdd4943ca445d5647db2/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/firebase-iid-interop-16.0.0.aar/16b522a0ee0855a265a3dbb7eb93cbf3/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/play-services-base-15.0.1.aar/fa2793b6dae0760211be3337d45490a8/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/play-services-tasks-15.0.1.aar/632221d2b248891505de8d7dff0d0822/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/firebase-measurement-connector-16.0.0.aar/e7b2006f55699a7b4cdd12b34f5beb01/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/play-services-stats-15.0.1.aar/b1a3836fff7c5afc91593e4d69c553c5/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/play-services-basement-15.0.1.aar/5400c9b61c30fddb1fdc8a6334231ed2/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/zxing-1.9.8.aar/c331b0523fd6201ae868eb04e43bdd7d/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/core-1.9.8.aar/30c3f3d9ca7290a1eca96eb98727ec20/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-v4-27.1.1.aar/a730d8fe960c7a51bb1ceeea121302dc/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/modules-2/files-2.1/org.jetbrains.kotlin/kotlin-stdlib-jdk7/1.2.51/a37c0066df119c02f034bf17159089b588ccc41a/kotlin-stdlib-jdk7-1.2.51.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/customtabs-27.1.1.aar/82c819ee5dc45f71cca7e0b31d9f25fc/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-media-compat-27.1.1.aar/804f32e3f17e680bd8a49797884e390f/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/appcompat-v7-27.1.1.aar/02a801060b8f6b2f9e802073865410dd/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-fragment-27.1.1.aar/72757299e459d3f6e3957fb2ff812ba3/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/recyclerview-v7-27.1.1.aar/e6f5880f875cd19a175c1f0a9dbc6b83/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/animated-vector-drawable-27.1.1.aar/91a6562376378503363470716b36ffc3/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-core-ui-27.1.1.aar/edbb889b6c3a550b5d2c15531b41f01f/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-core-utils-27.1.1.aar/8a5babd03c935863f76d26cc2d3056c2/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/transition-27.1.1.aar/7e6a4117a37b9bce31739e91091f8420/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-vector-drawable-27.1.1.aar/cdb1e8790c771bcf63aee84d31b360a5/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/support-compat-27.1.1.aar/5050d568643ebdf811aecd0a01539cb6/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/modules-2/files-2.1/com.android.support/support-annotations/27.1.1/39ded76b5e1ce1c5b2688e1d25cdc20ecee32007/support-annotations-27.1.1.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/modules-2/files-2.1/org.checkerframework/checker-qual/2.5.0/f03b9f07c2b002f35f81338e6cde3fc191b91951/checker-qual-2.5.0.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/versionedparcelable-1.0.0.aar/ab8e36f85714055d1049f02b85382186/jars/classes.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/modules-2/files-2.1/androidx.collection/collection/1.0.0/42858b26cafdaa69b6149f45dfc2894007bc2c7a/collection-1.0.0.jar] (filtered)
Reading program jar [/home/chandu/.gradle/caches/transforms-1/files-1.1/lifecycle-runtime-2.0.0.aar/808cb94913583105dfbe3b1ff3eb61d6/jars/classes.jar] (filtered)
[... cut ...]
    • All Android licenses accepted.

[✓] Android Studio (version 3.3)
    • Android Studio at /opt/android-studio
    • Flutter plugin version 33.3.1
    • Dart plugin version 182.5215
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1248-b01)

[✓] VS Code (version 1.31.1)
    • VS Code at /usr/share/code
    • Flutter extension version 2.24.0

[✓] Connected device (1 available)
    • Redmi Note 4 • 192.168.0.100:5555 • android-arm64 • Android 7.0 (API 24)

• No issues found!
```

FWeissenb, 2019-03-29:
Same here

`FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.
> java.lang.RuntimeException: java.lang.RuntimeException: com.android.builder.dexing.DexArchiveMergerException: Unable to merge dex
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
* Get more help at https://help.gradle.org
BUILD FAILED in 7s`

FWeissenb, 2019-03-29:
Using 0.4.2 "only" throws deprecated  notices.

efortuna, 2019-06-26:
FWIW, I followed these instructions in my app: https://github.com/flutter/flutter/issues/23131#issuecomment-475907260 and was able to get it to run.

pauldemarco, 2019-10-18:
Please refer to https://github.com/flutter/flutter/issues/23131#issuecomment-475907260 for the solution if you come across this. Closing for now.

closed 2019-10-18 by pauldemarco