pauldemarco/flutter_blue #200: Update protobuf dependency

artob, 2019-02-28:
As it stands, this plugin is unusable in apps that rely on the current [protobuf](https://pub.dartlang.org/packages/protobuf) version:

    $ flutter packages get
    Because flutter_blue >=0.4.2 depends on protobuf ^0.10.5 and myapp depends on protobuf ^0.13.4, flutter_blue >=0.4.2 is forbidden.
    So, because myapp depends on flutter_blue ^0.5.0, version solving failed.
    Running "flutter packages get" in myapp...
    pub get failed (1)

Given that this isn't the first time this issue has surfaced (https://github.com/pauldemarco/flutter_blue/issues/133, https://github.com/pauldemarco/flutter_blue/pull/135, etc), perhaps it might be a notion to not only update protobuf, but to also going forward relax the strictness of the version dependency somewhat?

RoboEvangelist, 2019-04-18:
My phone has protobuf 3.0.0 shared libraries in the /system/libs/ path when I use adb shell. How can I upgrade protobuf to 3.6.1? Can I compile on my ubuntu computer using android toolchain, and then just transfer and replace the libraries inside my phone?

pauldemarco, 2019-06-19:
This should be resolved in the latest version of the plugin.

closed 2019-06-19 by pauldemarco