pauldemarco/flutter_blue #201: Unhandled error in method connect() of class FlutterBlue

gordonwongcherrypicks, 2019-03-01:
When I try to connect to a device with an iphone, the following error may happen randomly:

```
 [VERBOSE-2:shell.cc(184)] Dart Error: Unhandled exception:
PlatformException(connect, Peripheral not found, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:551:7)
#1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:292:18)
<asynchronous suspension>
#2      FlutterBlue.connect (file:///Users/gordonwong/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/lib/src/flutter_blue.dart:131:22)
<asynchronous suspension>
```

I suggest adding a try-catch block inside method **connect()** of class **FlutterBlue** so the error can be  handled:
```
try {
      await _channel.invokeMethod('connect', request.writeToBuffer());
    }catch(e){
      throw e;
    }
```

ericstumper, 2019-10-29:
I am also running into this occasionally using flutter_blue 0.6.3+1. We are trying to run the connect process zoned but are still not able to catch this error and react accordingly.

yiguozhen0510, 2020-09-04:
I also have this problem,hava you soleve it ?