pauldemarco/flutter_blue #204: How to find out if we're already connected to a device?

lambasoft, 2019-03-05:
When my app start, I need to know if I'm already connected to a device. How can I do that using flutter_blue, I couldn't find the function anywhere.

Fleximex, 2019-03-12:
What I do is keep track of the connected devices in the Flutter app itself. So when you are connected you just set a boolean to true.

Even better is to use this (also used in the Flutter Blue example):
```
deviceStateSubscription = device.onStateChanged().listen((s) {
  setState(() {
    deviceState = s;
  });
});
```
This will set deviceState to either of 4 states.

Then simply write a isConnected bool in your code:
`bool get isConnected => (deviceState == BluetoothDeviceState.connected);`

When the device you are connected to disconnects the `deviceState` should automatically be updated by the `onStateChanged()` stream.
When your Flutter app disconnects make sure you hard code `deviceState = BluetoothDeviceState.disconnected;` in your code. When you disconnect the `device.state.then` part may never be called again because `device` may already be null or the `deviceStateSubscription` is already cancelled or null (depending on your code).

insequent, 2019-04-24:
As far as I can tell, the only way to know about connected devices is to keep track of it from the `.connect()` on? I don't see any way to pull already connected devices, via FlutterBlue or the platform. `.scan()` seems to skip connected ones as well.

rrnicolay, 2019-06-22:
I'm needing this too.
Using version 0.6.0+1.
The connect() function returns void. How to check if the connection succeded?

LastMonopoly, 2019-06-23:
try FlutterBlue.instance.connectedDevices

rrnicolay, 2019-06-24:
I'm using the code below. But, it would be better if connect() returned the status.

```
void _connect(BluetoothDevice device) async {

    _stopScan();

    await device.connect();

    List<BluetoothDevice> connectedDevices = await flutterBlue.connectedDevices;

    if(connectedDevices.contains(device)) {
      Navigator.pushReplacementNamed(context, '/home');
    }
  }
```

enricobenedos, 2020-01-08:
I've two devices connected to my Galaxy Tab S3 but using @LastMonopoly snippet the connected devices list is always empty even if it is not the truth
[image]
[image]

LastMonopoly, 2020-01-08:
@enricobenedos FlutterBlue.instance.connectedDevices will give u devices connected using FlutterBlue, not all Bluetooth-connected devices

enricobenedos, 2020-01-08:
@LastMonopoly thank you for the clarification!

J0nSnuw, 2020-04-14:
So just to be clear, there is no way to get a list of the already connected devices to the phone prior to FlutterBlue being initialized? I am curious about what the limitations are that are stopping this from happening.

themobilecoder, 2020-04-30:
Just wondering as well if there is a way to know currently connected device **NOT** connected using flutter_blue.

Ahmadre, 2020-07-11:
I have the same issue! For security reasons there're bluetooth devices that don't broadcast themselves, but only connect to devices and don't let anyone find that device.

SaikCaskey, 2022-08-27:
Some day I want to look into this, I don't understand why it isn't what we expect as I see in Logcat that android is correctly reporting the connected devices. I also don't understand why people aren't crying about this more, it's kind of a big deal if you actually need to connect to a device for your usecase

ngoan98tv, 2023-05-30:
No solution for years?!
hmm, I also looking for it, my app need to listen if the phone auto connect to a device (which already connected before).

shubhamsinghmutualmobile, 2023-07-03:
Maybe it's worthwhile to raise this same ticket in the [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus) repository as it is actively being maintained by the community since `flutter_blue` is not.