pauldemarco/flutter_blue #205: Cant find any device

MuhammadElhrz, 2019-03-05:
I am using oppo f5 for connecting to heart rate monitor polar h10 but the mobile cant find any devices while scanning it doesnt even go to the listen function in the code
What is the problem?

thangd2it, 2019-03-11:
Same me, I use iphone 6 to scanning but nothing device detected. What went wrong ???

Fleximex, 2019-03-12:
Give us some more info. Normally Flutter Blue will find a lot of (nameless) devices. Did you try the example?

RiteshJariwala, 2019-03-20:
I am also getting same thing. No device found. anybody care to help with it?

RiteshJariwala, 2019-03-20:
Ok, i got it now. It is required to "ON" location services.

ycktw, 2019-03-21:
I've also found the same requirement on non flutter blue apps. It seems not
flutter or flutter blue's issue. And sometimes, it is required to disable
and enable bluetooth.

heinhtetaung92, 2019-09-26:
Thanks @RiteshJariwala . It solves the issue. By any chance, do u know why they need location service to discover devices and why it's not optional.