pauldemarco/flutter_blue #206: Does this Plugin support Android X now?

100cm, 2019-03-06:
As the title said ,i don't know if this plugin support android X ,someone could help me?

lemonbuzz, 2019-03-10:
Why would it not support android x

markflarup, 2019-03-15:
I doubt it. The plugin supports AndroidX, but it might be due to the fact that some of the packages `flutter_blue` depends on doesn't support AndroidX.  
Based on my experience I can't successfully build my app for Android. Adding `flutter_blue` as a dependency results in `Gradle task assembleDevDebug failed with exit code 1` when I run the program. However, I can successfully build the same code base for iOS. 

### Log
`flutter run --verbose` returns the following:
```
[  +43 ms] executing: [/Users/[USERNAME]/Library/flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[  +50 ms] Exit code 0 from: git rev-parse --abbrev-ref --symbolic @{u}
[        ] origin/beta
[        ] executing: [/Users/[USERNAME]/Library/flutter/] git rev-parse --abbrev-ref HEAD
[  +11 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] beta
[        ] executing: [/Users/[USERNAME]/Library/flutter/] git ls-remote --get-url origin
[  +11 ms] Exit code 0 from: git ls-remote --get-url origin
[        ] https://github.com/flutter/flutter.git
[   +2 ms] executing: [/Users/[USERNAME]/Library/flutter/] git log -n 1 --pretty=format:%H
[  +17 ms] Exit code 0 from: git log -n 1 --pretty=format:%H
[        ] 8661d8aecd626f7f57ccbcb735553edc05a2e713
[        ] executing: [/Users/[USERNAME]/Library/flutter/] git log -n 1 --pretty=format:%ar
[  +12 ms] Exit code 0 from: git log -n 1 --pretty=format:%ar
[        ] 4 weeks ago
[   +1 ms] executing: [/Users/[USERNAME]/Library/flutter/] git describe --match v*.*.* --first-parent --long --tags
[  +67 ms] Exit code 0 from: git describe --match v*.*.* --first-parent --long --tags
[        ] v1.2.1-0-g8661d8aec
[ +491 ms] executing: /Users/[USERNAME]/Library/Android/sdk/platform-tools/adb devices -l
[  +10 ms] Exit code 0 from: /Users/[USERNAME]/Library/Android/sdk/platform-tools/adb devices -l
[        ] List of devices attached
           13ad4c7f               device usb:487657728X product:OnePlus3 model:ONEPLUS_A3003 device:OnePlus3 transport_id:1
[  +13 ms] executing: idevice_id -h
[ +174 ms] executing: which ideviceinstaller
[   +6 ms] Exit code 0 from: which ideviceinstaller
[        ] /usr/local/bin/ideviceinstaller
[        ] executing: which iproxy
[   +6 ms] Exit code 0 from: which iproxy
[        ] /usr/local/bin/iproxy
[   +4 ms] /usr/bin/xcrun simctl list --json devices
[ +449 ms] Found plugin cloud_firestore at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/cloud_firestore-0.9.5+2/
[  +13 ms] Found plugin firebase_auth at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_auth-0.8.1+3/
[   +2 ms] Found plugin firebase_core at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_core-0.3.1+1/
[   +3 ms] Found plugin firebase_messaging at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_messaging-4.0.0+1/
[  +11 ms] Found plugin flutter_blue at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/
[  +41 ms] Found plugin shared_preferences at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/shared_preferences-0.5.1+1/
[  +43 ms] Found plugin cloud_firestore at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/cloud_firestore-0.9.5+2/
[   +7 ms] Found plugin firebase_auth at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_auth-0.8.1+3/
[   +1 ms] Found plugin firebase_core at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_core-0.3.1+1/
[   +2 ms] Found plugin firebase_messaging at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/firebase_messaging-4.0.0+1/
[   +6 ms] Found plugin flutter_blue at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/
[  +23 ms] Found plugin shared_preferences at /Users/[USERNAME]/Library/flutter/.pub-cache/hosted/pub.dartlang.org/shared_preferences-0.5.1+1/
[  +55 ms] /Users/[USERNAME]/Library/Android/sdk/platform-tools/adb -s 13ad4c7f shell getprop
[  +63 ms] ro.hardware = qcom
[+1349 ms] Launching lib/main.dart on ONEPLUS A3003 in debug mode...
[  +24 ms] Initializing gradle...
[   +7 ms] Using gradle from /Users/[USERNAME]/GitHub/[PROJECT]/android/gradlew.
[+1567 ms] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/test/' (OS Error: Permission denied, errno = 13)
[        ] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/phpmyadmin/' (OS Error: Permission denied, errno = 13)
[        ] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/cdcol/' (OS Error: Permission denied, errno = 13)
[        ] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/mysql/' (OS Error: Permission denied, errno = 13)
[        ] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/Assignment2/' (OS Error: Permission denied, errno = 13)
[        ] Exception while looking for Android Studio: FileSystemException: Directory listing failed, path = '/Applications/XAMPP/xamppfiles/var/mysql/performance_schema/' (OS Error: Permission denied, [... cut ...]
[        ]          *********************************************************
[        ] 60 actionable tasks: 1 executed, 59 up-to-date
[ +404 ms] Running Gradle task 'assembleDevDebug'... (completed in 4.0s)
[   +5 ms] "flutter run" took 14,614ms.
[        ] "flutter run" took 14,614ms.
[        ] Gradle task assembleDevDebug failed with exit code 1
```

Besides the rather generic error message I guess the line `[        ] > Task :flutter_blue:generateDebugProto FAILED` has something to do with it?

pauldemarco, 2019-03-15:
I'll look into this today.

On Fri, Mar 15, 2019, 2:11 AM markflarup <notifications@github.com> wrote:

> I doubt it. The plugin supports AndroidX, but it might be due to the fact
> that some of the packages flutter_blue depends on doesn't support
> AndroidX.
> Based on my experience I can't successfully [...]

markflarup, 2019-03-26:
> I'll look into this today.

Have you had the time to look into the problem? 😊

kushtra24, 2019-04-04:
Having the same issue, cant add it to the pubspec.yaml without breaking:

`
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:preDebugBuild'.
> Android dependency 'com.android.support:support-compat' has different version for the compile (26.1.0) and runtime (27.1.1) classpath. You should manually set the same version via DependencyResolution

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
Finished with error: Gradle task assembleDebug failed with exit code 1
*******************************************************************************************
The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app.
See https://goo.gl/CP92wY for more information on the problem and how to fix it.
*******************************************************************************************
`

markflarup, 2019-04-08:
> ```
>   * What went wrong:
>   Execution failed for task ':app:preDebugBuild'.
>   > Android dependency 'com.android.support:support-compat' has different version for the compile ( 
>   26.1.0) and runtime (27.1.1) classpath. You should manually set the same version via 
>   DependencyResolution
> ```

Have you tried to update `Android dependency 'com.android.support:support-compat'` through DependencyResolution as suggested by the log? Simply to see if a new error will surface or what will happen 😊

mikes222, 2019-05-04:
[        ] > Task :flutter_blue:compileDebugRenderscript NO-SOURCE
    [        ] > Task :flutter_blue:generateDebugBuildConfig UP-TO-DATE
    [        ] > Task :flutter_blue:extractDebugProto UP-TO-DATE
    [        ] > Task :flutter_blue:extractProto UP-TO-DATE
    [   +1 ms] > Task :flutter_blue:extractIncludeDebugProto UP-TO-DATE
    [   +1 ms] > Task :flutter_blue:generateDebugProto FAILED
    [   +1 ms]          *********************************************************
    [   +2 ms] WARNING: This version of shared_preferences will break your Android build if it or its  dependencies aren't
    compatible with AndroidX.
    [   +1 ms]          See https://goo.gl/CP92wY for more information on the problem and how to fix it.
    [   +1 ms]          This warning prints for all Android build failures. The real root cause of the error may be unrelated.
    [   +1 ms]          *********************************************************

I do have similar problems with that library with the most current version (0.5.0). Found the above message when compiling in the terminal with `flutter run -v`. I need to disable the whole library for now until the issue is fixed since I am unable to start the app for android.

markflarup, 2019-05-06:
> I do have similar problems with that library with the most current version (0.5.0). Found the above message when compiling in the terminal with `flutter run -v`. I need to disable the whole library for now until the issue is fixed since I am unable to start the app for android.

Try pulling this PR (#245) where I have added a dependency and updated the versions needed. It works for me. Let me know if this works for others.
You can add local package by specifying the path to the folder containing a `.yaml` file, e.g.:

```
  flutter_blue:
    path: ../fork/flutter_blue/
```

Skquark, 2019-05-13:
Also ran into this issue (using flutter_blue_beacon) with my AndroidX migrated project.  With Android Studio, right click android folder, then Flutter->Open Android Module in Android Studio and after it scans everything (had already gone through Refactor->Migrate to AndroidX..), it stops at this error:
```
org.gradle.api.tasks.TaskValidationException: A problem was found with the configuration of task ':flutter_blue:generateDebugProto'.
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.report(ValidatingTaskExecuter.java:66)
...
	at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:55)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.gradle.api.InvalidUserDataException: Directory 'C:\Projectz\myproject\build\flutter_blue\extracted-include-protos\main' specified for property '$3' does not exist
```
Maybe that's more helpful to you than it is for me.. I haven't tried your PR yet, so that might fix it.  Should I wait for release or point it to your update?  Thanks...

markflarup, 2019-05-13:
> Maybe that's more helpful to you than it is for me.. I haven't tried your PR yet, so that might fix it. Should I wait for release or point it to your update? Thanks...

I would like you to try the pull request for two reasons: 
1) it increases the validity of the PR, 
2) I don't know when @pauldemarco have time to test and eventually accept the pull request into the master branch.

Skquark, 2019-05-14:
Thanks for providing the yaml git ref part, had difficulties figuring that out.  Anyways, I got that in place, cleaned all the caches and tried again, but it was still giving same problem.. Tried a few things, but it still fails on the generateDebugProto task.  Good to have all those dependancies updated, but might be something else gumming up the code..

pauldemarco, 2019-06-19:
Please try the latest version of the plugin and re-open if the issue persists.

closed 2019-06-19 by pauldemarco