pauldemarco/flutter_blue #207: can't connect to thermal bluetooth printer

erwintan96, 2019-03-06:
i am trying to run your example and i can't connect to my thermal bluetooth printer. 

I got this output and i dont know how to fix it

D/BluetoothGatt(22380): connect() - device: 0F:02:17:B0:24:08, auto: true
D/BluetoothGatt(22380): registerApp()
D/BluetoothGatt(22380): registerApp() - UUID=2b627454-9683-4285-aef0-7d22433ad124
D/BluetoothGatt(22380): onClientRegistered() - status=0 clientIf=8
I/flutter (22380): localName: BlueTooth Printer
I/flutter (22380): manufacturerData: {}
I/flutter (22380): serviceData: {}
I/flutter (22380): localName: BlueTooth Printer
I/flutter (22380): manufacturerData: {}
I/flutter (22380): serviceData: {}
D/BluetoothGatt(22380): onClientConnectionState() - status=133 clientIf=8 device=0F:02:17:B0:24:08
D/FlutterBluePlugin(22380): [onConnectionStateChange] status: 133 newState: 0
D/BluetoothAdapter(22380): STATE_ON
D/BluetoothManager(22380): getConnectionState()
D/BluetoothManager(22380): getConnectedDevices
I/flutter (22380): State refreshed: BluetoothDeviceState.disconnected
D/BluetoothGatt(22380): close()
D/BluetoothGatt(22380): unregisterApp() - mClientIf=8
Lost connection to device.

could you give me some advice?

Fleximex, 2019-03-12:
Download the app 'BLE Scanner' for either Android or iOS. The app will give you all the information of the Bluetooth device in question and the ability to connect.
https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner
https://itunes.apple.com/us/app/id1221763603

Check if you can connect with the device that way.

shajeeck, 2019-04-02:
@erwintan96 where u able you able to print using bluetooth printer in flutter?

bretie, 2019-07-18:
Fork your own version of the library. Under the "connect" case change the following...

`
                // New request, connect and add gattServer to Map
                BluetoothGatt gattServer;
                if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
                    gattServer = device.connectGatt(activity, options.getAndroidAutoConnect(), mGattCallback, BluetoothDevice.TRANSPORT_LE);
                } else {
                    gattServer = device.connectGatt(activity, options.getAndroidAutoConnect(), mGattCallback);
                }
                mGattServers.put(deviceId, gattServer);
                result.success(null);
                break;
`

to 

`
                // New request, connect and add gattServer to Map
                BluetoothGatt gattServer;
                gattServer = device.connectGatt(activity, options.getAndroidAutoConnect(), mGattCallback, BluetoothDevice.TRANSPORT_LE);
                mGattServers.put(deviceId, gattServer);
                result.success(null);
                break;
`

Device now connects. I believe if you change your minSDK version or something of the sorts it would work. But this SDK check is breaking it for me on a MotoG3

rituuniversal, 2024-08-30:
no printer connected show the message how to connect the device in my application