pauldemarco/flutter_blue #209: Can't compile after regenerate dart proto

thachnb85, 2019-03-08:
1. Issue:
Can't compile after regenerating dart code from protoc

2. Procedure
- Adding new message in flutterblue.proto
```
message AdvertisingSettings {
  string local_name = 1;
  string service_uuid = 2;
}
```

- Get latest protoc and dart-protoc-plugin
- Now just run project, we can see bunch of errors:
```
Launching lib/main.dart on Air 1 LTE in debug mode...
Automatically signing iOS for device deployment using specified development team in Xcode project: 123123123
Running Xcode build...
Xcode build done.                                            5.2s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    === BUILD TARGET Runner OF PROJECT Runner WITH CONFIGURATION Debug ===
    The use of Swift 3 @objc inference in Swift 4 mode is deprecated. Please address deprecated @objc inference warnings, test your code with “Use of deprecated Swift 3 @objc inference” logging enabled, and then disable inference by changing the "Swift 3 @objc Inference" build setting to "Default" for the "Runner" target.
    === BUILD TARGET Runner OF PROJECT Runner WITH CONFIGURATION Debug ===

    Compiler message:
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:69:24: Error: Too many positional arguments: 7 allowed, but 8 found.
    Try removing the extra positional arguments.
        ..m<int, List<int>>(4, 'manufacturerData', 'AdvertisementData.ManufacturerDataEntry',$pb.PbFieldType.O3, $pb.PbFieldType.OY, null, null, null )
                           ^
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:70:27: Error: Too many positional arguments: 7 allowed, but 8 found.
    Try removing the extra positional arguments.
        ..m<String, List<int>>(5, 'serviceData', 'AdvertisementData.ServiceDataEntry',$pb.PbFieldType.OS, $pb.PbFieldType.OY, null, null, null )
                              ^
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:273:7: Error: The method 'pc' isn't defined for the class 'BuilderInfo'.
     - 'BuilderInfo' is from 'package:protobuf/protobuf.dart' ('file:///Users/user/.pub-cache/hosted/pub.dartlang.org/protobuf-0.10.8/lib/protobuf.dart').
    Try correcting the name to the name of an existing method, or defining a method named 'pc'.
        ..pc<BluetoothCharacteristic>(4, 'characteristics', $pb.PbFieldType.PM,BluetoothCharacteristic.create)
          ^^
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:274:7: Error: The method 'pc' isn't defined for the class 'BuilderInfo'.
     - 'BuilderInfo' is from 'package:protobuf/protobuf.dart' ('file:///Users/user/.pub-cache/hosted/pub.dartlang.org/protobuf-0.10.8/lib/protobuf.dart').
    Try correcting the name to the name of an existing method, or defining a method named 'pc'.
        ..pc<BluetoothService>(5, 'includedServices', $pb.PbFieldType.PM,BluetoothService.create)
          ^^
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:315:7: Error: The method 'pc' isn't defined for the class 'BuilderInfo'.
     - 'BuilderInfo' is from 'package:protobuf/protobuf.dart' ('file:///Users/user/.pub-cache/hosted/pub.dartlang.org/protobuf-0.10.8/lib/protobuf.dart').
    Try correcting the name to the name of an existing method, or defining a method named 'pc'.
        ..pc<BluetoothDescriptor>(4, 'descriptors', $pb.PbFieldType.PM,BluetoothDescriptor.create)
          ^^
    flutter_blue-0.5.0/lib/gen/flutterblue.pb.dart:484:7: Error: The method 'pc' isn't defined for the class 'BuilderInfo'.
     - 'BuilderInfo' is from 'package:protobuf/protobuf.dart' ('file:///Users/user/.pub-cache/hosted/pub.dartlang.org/protobuf-0.10.8/lib/protobuf.dart').
    Try correcting the name to the name of an existing method, or defining a method named 'pc'.
        ..pc<BluetoothService>(2, 'services', $pb.PbFieldType.PM,BluetoothService.create)
          ^^
    Compiler failed on /Users/user/Project_Flutter/ble_poc/lib/main.dart
    Failed to package /Users/user/Project_Flutter/ble_poc.

Could not build the precompiled application for the device.

Error launching application on Air 1 LTE.

```

Any idea how to resolve this issue? Thank you.

thachnb85, 2019-03-08:
Tried with flutterblue.proto
```
protoc --dart_out=../lib/gen ./flutterblue.proto
```
Then compile & run, it is still showing the same issue above.

artob, 2019-03-10:
Perhaps related to #200. You might need to use an older dart-protoc-plugin version.

thachnb85, 2019-03-11:
Thanks @artob  Which version are you using now?

pauldemarco, 2019-06-19:
Please try the latest version of the plugin and re-open if the issue persists.

closed 2019-06-19 by pauldemarco