pauldemarco/flutter_blue #212: Impossible to add package in pubspec.yaml

nitneuq33, 2019-03-10:
Hello, I tried to add flutter ble packages in my pubspec.yaml. But I have this message :

```
Because every version of flutter_test from sdk depends on convert 2.0.2 and flutter_blue >=0.4.2 depends on convert ^2.1.1, flutter_test from sdk is incompatible with flutter_blue >=0.4.2.
So, because flutter_app_ble_1 depends on both flutter_blue ^0.5.0 and flutter_test any from sdk, version solving failed.
```

I tried to downgrade the package, tried to migrate in androidX, tried, tried to change min sdk to 18 and compile sdk to 28.

artob, 2019-03-10:
Remove or override your [flutter_test](https://pub.dartlang.org/packages?q=flutter_test) dependency in your `pubspec.yaml`.

pauldemarco, 2019-06-19:
Please try the latest version of the plugin and re-open if the issue persists.

closed 2019-06-19 by pauldemarco

taf2, 2020-12-01:
Super new to flutter and running into this issue as well here is my pubspec.yaml:

```
name: clockapp
description: A new Flutter project.

# The following defines the version and build number for your application.
# A version number is three numbers separated by dots, like 1.2.43
# followed by an optional build number separated by a +.
# Both the version and the builder number may be overridden in flutter
# build by specifying --build-name and --build-number, respectively.
# In Android, build-name is used as versionName while build-number used as versionCode.
# Read more about Android versioning at https://developer.android.com/studio/publish/versioning
# In iOS, build-name is used as CFBundleShortVersionString while build-number used as CFBundleVersion.
# Read more about iOS versioning at
# https://developer.apple.com/library/archive/documentation/General/Reference/InfoPlistKeyReference/Articles/CoreFoundationKeys.html
version: 1.0.0+1

environment:
  sdk: ">=2.1.0 <3.0.0"

dependencies:
  flutter:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^0.1.2
  flutter_blue: ^0.7.3

dev_dependencies:
  flutter_test:
    sdk: flutter

# For information on the generic Dart part of this file, see the
# following page: https://dart.dev/tools/pub/pubspec

# The following section is specific to Flutter.
flutter:

  # The following line ensures that the Material Icons font is
  # included with your application, so that you can use the icons in
  # the material Icons class.
  uses-material-design: true

  # To add assets to your application, add an assets section, like this:
  # assets:
  #   - images/a_dot_burr.jpeg
  #   - images/a_dot_ham.jpeg

  # An image asset can refer to one or more resolution-specific "variants", see
  # https://flutter.dev/assets-and-images/#resolution-aware.

  # For details regarding adding assets from package dependencies, see
  # https://flutter.dev/assets-and-images/#from-packages

  # To add custom fonts to your application, add a fonts section here,
  # in this "flutter" section. Each entry in this list should have a
  # "family" key with the font family name, and a "fonts" key with a
  # list giving the asset and other descriptors for the font. For
  # example:
  # fonts:
  #   - family: Schyler
  #     fonts:
  #       - asset: fonts/Schyler-Regular.ttf
  #       - asset: fonts/Schyler-Italic.ttf
  #         style: italic
  #   - family: Trajan Pro
  #     fonts:
  #       - asset: fonts/TrajanPro.ttf
  #       - asset: fonts/TrajanPro_Bold.ttf
  #         weight: 700
  #
  # For details regarding fonts from package dependencies,
  # see https://flutter.dev/custom-fonts/#from-packages
```

Here is the output from running ```flutter pub get```
```
Because every version of flutter_test from sdk depends on meta 1.1.8 and flutter_blue >=0.7.3 depends on meta ^1.2.3, flutter_test from sdk is incompatible with flutter_blue >=0.7.3.
So, because clockapp depends on both flutter_blue ^0.7.3 and flutter_test any from sdk, version solving failed.
Running "flutter pub get" in clockapp...
pub get failed (1; So, because clockapp depends on both flutter_blue ^0.7.3 and flutter_test any from sdk, version solving
failed.)
```

Sorry I am not sure how to override flutter_test... any tips?

taf2, 2020-12-01:
I've tried commenting out 

```
dev_dependencies:
  flutter_test:
    sdk: flutter
```

now I get the following:
```
flutter pub get
Because flutter_blue >=0.7.3 depends on meta ^1.2.3 and every version of flutter from sdk depends on meta 1.1.8, flutter_blue >=0.7.3 is incompatible with flutter from sdk.
So, because clockapp depends on both flutter any from sdk and flutter_blue ^0.7.3, version solving failed.
Running "flutter pub get" in clockapp...
pub get failed (1; So, because clockapp depends on both flutter any from sdk and flutter_blue ^0.7.3, version solving failed.)
```

taf2, 2020-12-01:
Okay, I think I made progress turns out I had an old version of flutter checked out... I figured this out initially by running:

```
flutter upgrade
```

Then I navigated to the reported directory and ran:
```
git co master
git pull --rebase origin master
```

back into flutter project and 
```
flutter pub get
```

Things still no go on master so then I realized:
```
flutter channel stable
flutter upgrade
```

And now we're operational - thank you so much for the great BLE library!  Hopefully my nub stumbling helps someone else...