pauldemarco/flutter_blue #213: 2 characteristics with the same UUID

truongsinh, 2019-03-11:
The device I'm working on has 2 characteristics with the same UUID, but different properties (permissions), one for "write" and one for "notify".

The rationale of the manufacturer as far as I know, is that these "characteristics" are the same, the one with "write" permission is used to initiate the action from mobile, while the one with "notify" is used to notify mobile when that action is initiated on the device itself.

Reading flutter_blue source code, I see that they are locating by characteristicId only (together with serviceId and secondaryServiceId)

truongsinh, 2019-03-11:
Reference: https://stackoverflow.com/questions/27413272/what-gatt-services-use-multiple-characteristics-with-the-same-uuid

truongsinh, 2019-03-11:
PR https://github.com/pauldemarco/flutter_blue/pull/215

pauldemarco, 2019-03-13:
The device you are interfacing with has duplicate characteristics in the same service?

truongsinh, 2019-03-13:
That's correct, as you can read from the the reference link I put above. 

> A service may have multiple characteristic definitions with the same Characteristic UUID. (Vol 3, Part G, 3.3.1 "Characteristic Declaration")

> Returns the instance ID for this characteristic.
>
> If a remote device offers multiple characteristics with the same UUID, the instance ID is used to distuinguish between characteristics.

pauldemarco, 2019-03-13:
I do see that in the specification, a single line from section 3.3.1.
If it's in the spec, FlutterBlue must be able to handle it. I'll just need to verify it's been adopted on both Android and iOS, and in what way.

Thanks for bringing this to my attention.

DanielPiflaks, 2021-08-15:
I'm facing the same issue.
I can see that @truongsinh indeed fixed it in https://github.com/pauldemarco/flutter_blue/pull/215 but I can't find this fix on the master or on 0.8v.
Am I missing something?

mgeilich, 2023-02-28:
I also have this same issue.
Is there a fix?

wlodzix, 2023-04-11:
Any updates here?