pauldemarco/flutter_blue #214: Can flutter blue run on BT5 ?

ycktw, 2019-03-11:
Dear Sirs,

I write codes for devices with BT, which are mostly toys for kids.
I am now testing if flutter blue can work with BT5.

My test device chip is RTL8762, my phone is iPhone 8.
flutter blue: ec012a32f532ef9e8e4f8eeb0d21c58f5936bec8
flutter: v1.3.4

It has successfully built up connection, and data received.
However, the speed is very slow than my expectation. (only 20 ~ 38k)

Could you give me some advises to go forward ?

Thanks for your further comments,
yck

Fleximex, 2019-03-12:
It is not the question whether Flutter Blue can work with Bluetooth 'version number'. Bluetooth comes in two flavors: Classic and Low Energy. Any Bluetooth device supports Classic. Bluetooth devices with version 4.0 and higher also support Low Energy.
Flutter Blue currently only supports Low Energy, thus only works with devices with Bluetooth 4.0 or higher chips.

What are you trying to transfer for it to exceed 38kbps?

Some reading on BLE speeds: https://devzone.nordicsemi.com/f/nordic-q-a/1105/how-do-i-calculate-throughput-for-a-ble-link/5185#5185

ycktw, 2019-06-03:
I got compared the throughput report with Realtek's iOS (ObjectC) code.
And the conclusion is Positive --
iOS's speed on BT5 is originally lower than Android's.

closed 2019-06-03 by ycktw