pauldemarco/flutter_blue #221: "Random" Disconnects (didDisconnectPeripheral) in iOS

mamen2022, 2019-03-21:
Hi,
I am developing a Flutter App for the past few months in Android and the Bluetooth is working correctly. Recently I began migrating to iOS and I keep getting "random" disconnects without any errors or crashes, besides this : 

```didDisconnectPeripheral```

Does anyone know why this happens or experienced the same problem?
I am testing with an iPhone 6 (iOS 12.1.4) and a ESP32 as ble central.
Any ideas ?

In comparison to Android, the bluetooth scan and connect seems at least 10 times slower, but the communication (write/read) between devices seems fine.

Best Regards

catalinio, 2019-04-05:
Hi,
I have also some different behaviours on flutter blue v0.5.0 between Android 8.0 (phone and tablet) and iOS 12.2 (iphone 5s, 8 and XR). My problems are in connection with ESP32 devices.
The scan, connection and communication (R/W) works fine on 2 services/characteristics (similarly fast on both Android and iOS), but problems are on disconnecting. On iOS the disconnect is not signalled to the esp32 side, but on Android it is.
I used `deviceConnection.cancel();` (and also `scanSubscription.cancel();`), in an explicit button, but when triggered it doesn't disconnect on iOS.
br

tharit-haup, 2019-10-29:
Is this close? What is the solution to this problem?

Ladegeraet, 2021-08-17:
Could someone find or solve the problem? I also have these complete random disconnects.

Lokendra16, 2023-03-28:
I am also facing the same issue.Unable to connect with ble device on Ios in android it is working fine. Getting this message
"didDisconnectPeripheral" If anyone know something about it then please let me know.