pauldemarco/flutter_blue #223: Issue with writeCharacteristic()

BradHacker, 2019-03-21:
I am trying to connect to an [Adafruit nrf8001](https://learn.adafruit.com/getting-started-with-the-nrf8001-bluefruit-le-breakout/) hooked up to an Arduino. It uses the UART service with the UUID `6E400001-B5A3-F393-E0A9-E50E24DCCA9E` as defined [here](https://learn.adafruit.com/getting-started-with-the-nrf8001-bluefruit-le-breakout/adding-app-support). I communicates with the TX and RX characteristics. I can connect to the device perfectly fine, but when I try to write to the TX characteristic, I get the error:

```
D/FlutterBluePlugin(20492): [onCharacteristicWrite] uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 128
E/flutter (20492): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: Exception: Failed to write the characteristic
E/flutter (20492): #0      BluetoothDevice.writeCharacteristic.<anonymous closure> 
    ../…/src/bluetooth_device.dart:132
... full output below
```

I set up my scanning for devices with:
```dart
setState(() {
  _scanning = true;
});
_scanSubscription =
  flutterBlue.scan(timeout: Duration(minutes: 1)).listen((scanResult) {
    if (scanResult.advertisementData.connectable) {
      setState(() {
        scanResults[scanResult.device.id] = scanResult;
      });
    }
  }, onDone: _cancelScanning);
```
and then I connect with:
```dart
setState(() {
  _deviceConnection = flutterBlue.connect(device).listen((s) {
    if (s == BluetoothDeviceState.connected) {
      print('Connected');
      setState(() {
        _device = device;
        _connected = true;
        _statusMessage = 'Connected to Cycler';
      });
      _cancelScanning();
      _discoverServices();
    } else {
      print('Not Connected');
      print(s.toString());
    }
  });
});
```
I then find the UART service and related characteristics as follows:

```dart
_discoverServices() {
  setState(() {
    _statusMessage = 'Discovering device services';
  });
  _device.discoverServices().then((s) {
    setState(() {
      _statusMessage = 'Services discovered';
      _deviceServices = s;
    });
    _findUART();
  });
}

_findUART() {
  if (_deviceServices != null && _deviceServices.isNotEmpty) {
    _deviceServices.forEach((service) {
      //servicesList.add(Text(service.uuid.toString()));
      if (service.uuid.toString().toLowerCase() ==
          '6E400001-B5A3-F393-E0A9-E50E24DCCA9E'.toLowerCase()) {
        print('UART found');
        _uartService = service;
        service.characteristics.forEach((c) {
          if (c.uuid.toString().toLowerCase() ==
              '6E400002-B5A3-F393-E0A9-E50E24DCCA9E'.toLowerCase()) {
            _tx = c;
            print('TX found');
          } else if (c.uuid.toString().toLowerCase() ==
              '6E400003-B5A3-F393-E0A9-E50E24DCCA9E'.toLowerCase()) {
            _rx = c;
            print('RX found');
          }
        });
      }
    });
  }
}
```

Then I have a button `onPressed:` linked to the method `_sendTest`

```dart
_sendTest() async {
  final value = await _device.writeCharacteristic(_tx, [0x12, 0x34],
      type: CharacteristicWriteType.withResponse);
  print(value);
}
```

Using Flutter 1.2.1. Running on Samsung Galaxy S8. Please let me know if you need any more relevant information.

Here's the full error:

```
E/flutter (20492): #0      BluetoothDevice.writeCharacteristic.<anonymous closure> 
../…/src/bluetooth_device.dart:132
E/flutter (20492): #1      _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #2      _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #3      _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
E/flutter (20492): #4      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
E/flutter (20492): #5      Future._propagateToListeners (dart:async/future_impl.dart:668:32)
E/flutter (20492): #6      Future._complete (dart:async/future_impl.dart:473:7)
E/flutter (20492): #7      _cancelAndValue (dart:async/stream_pipe.dart:63:12)
E/flutter (20492): #8      Stream.first.<anonymous closure> (dart:async/stream.dart:1170:11)
E/flutter (20492): #9      _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #10     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #11     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20492): #12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter (20492): #13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
E/flutter (20492): #14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
E/flutter (20492): #15     _WhereStream._handleData (dart:async/stream_pipe.dart:207:12)
E/flutter (20492): #16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
E/flutter (20492): #17     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #18     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #19     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20492): #20     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter (20492): #21     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
E/flutter (20492): #22     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
E/flutter (20492): #23     _MapStream._handleData (dart:async/stream_pipe.dart:232:10)
E/flutter (20492): #24     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
E/flutter (20492): #25     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #26     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #27     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20492): #28     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter (20492): #29     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
E/flutter (20492): #30     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
E/flutter (20492): #31     _MapStream._handleData (dart:async/stream_pipe.dart:232:10)
E/flutter (20492): #32     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
E/flutter (20492): #33     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #34     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #35     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20492): #36     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter (20492): #37     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
E/flutter (20492): #38     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
E/flutter (20492): #39     _WhereStream._handleData (dart:async/stream_pipe.dart:207:12)
E/flutter (20492): #40     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
E/flutter (20492): #41     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (20492): #42     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20492): #43     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20492): #44     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter (20492): #45     _DelayedData.perform (dart:async/stream_impl.dart:591:14)
E/flutter (20492): #46     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:707:11)
E/flutter (20492): #47     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:667:7)
E/flutter (20492): #48     _rootRun (dart:async/zone.dart:1120:38)
E/flutter (20492): #49     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter (20492): #50     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
E/flutter (20492): #51     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
E/flutter (20492): #52     _rootRun (dart:async/zone.dart:1124:13)
E/flutter (20492): #53     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter (20492): #54     _CustomZone.runGuarded (dart:async/zone.dar

okachobi, 2019-03-21:
Just guessing- but have you tried without CharacteristicWriteType.withResponse ?  I found that to be a problem before on iOS.

BradHacker, 2019-03-21:
I have tried that before.

earyzhe, 2019-03-24:
Hey, 

I'm also having a similar problem with writing a characteristic, but only on Android devices. 

iOS is working fine.

`
E/flutter ( 7662): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter ( 7662): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:564:7)
E/flutter ( 7662): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:302:33)
E/flutter ( 7662): <asynchronous suspension>
E/flutter ( 7662): #2      BluetoothDevice.writeCharacteristic (file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/lib/src/bluetooth_device.dart:114:10)
E/flutter ( 7662): <asynchronous suspension>
E/flutter ( 7662): #3      BluetoothInstance._turnSwitchOff (package:battery_saver/bluetooth_functions.dart:131:13)
E/flutter ( 7662): <asynchronous suspension>
E/flutter ( 7662): #4      BluetoothInstance.stopCharging (package:battery_saver/bluetooth_functions.dart:40:5)
E/flutter ( 7662): #5      MainBloc.batteryChargeChanged (package:battery_saver/bloc.dart:87:18)
E/flutter ( 7662): #6      _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter ( 7662): #7      _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter ( 7662): #8      _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter ( 7662): #9      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
E/flutter ( 7662): #10     _DelayedData.perform (dart:async/stream_impl.dart:591:14)
E/flutter ( 7662): #11     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:707:11)
E/flutter ( 7662): #12     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:667:7)
E/flutter ( 7662): #13     _rootRun (dart:async/zone.dart:1120:38)
E/flutter ( 7662): #14     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter ( 7662): #15     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
E/flutter ( 7662): #16     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
E/flutter ( 7662): #17     _rootRun (dart:async/zone.dart:1124:13)
E/flutter ( 7662): #18     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter ( 7662): #19     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
E/flutter ( 7662): #20     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
E/flutter ( 7662): #21     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
E/flutter ( 7662): #22     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5) 
`

BradHacker, 2019-03-25:
Hmm.... I can't find any documentation on the error code 128 I was getting. I've been looking for a few days. Anyone?

BradHacker, 2019-03-26:
On iOS with iPhone 7 i still get an error

```
didWriteValueForCharacteristic
[VERBOSE-2:ui_dart_state.cc(148)] Unhandled Exception: Exception: Failed to write the characteristic
#0      BluetoothDevice.writeCharacteristic.<anonymous closure> 
../…/src/bluetooth_device.dart:132
#1      _rootRunUnary (dart:async/zone.dart:1132:38)
#2      _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#3      _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
#4      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
#5      Future._propagateToListeners (dart:async/future_impl.dart:668:32)
#6      Future._complete (dart:async/future_impl.dart:473:7)
#7      _cancelAndValue (dart:async/stream_pipe.dart:63:12)
#8      Stream.first.<anonymous closure> (dart:async/stream.dart:1170:11)
#9      _rootRunUnary (dart:async/zone.dart:1132:38)
#10     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#11     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
#12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
#13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
#14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
#15     _WhereStream._handleData (dart:async/stream_pipe.dart:207:12)
#16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
#17     _rootRunUnary (dart:async/zone.dart:1132:38)
#18     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#19     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
#20     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
#21     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
#22     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
#23     _MapStream._handleData (dart:async/stream_pipe.dart:232:10)
#24     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
#25     _rootRunUnary (dart:async/zone.dart:1132:38)
#26     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#27     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
#28     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
#29     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
#30     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
#31     _MapStream._handleData (dart:async/stream_pipe.dart:232:10)
#32     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
#33     _rootRunUnary (dart:async/zone.dart:1132:38)
#34     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#35     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
#36     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
#37     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:263:7)
#38     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:132:11)
#39     _WhereStream._handleData (dart:async/stream_pipe.dart:207:12)
#40     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:164:13)
#41     _rootRunUnary (dart:async/zone.dart:1132:38)
#42     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#43     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
#44     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:336:11)
#45     _DelayedData.perform (dart:async/stream_impl.dart:591:14)
#46     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:707:11)
#47     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:667:7)
#48     _rootRun (dart:async/zone.dart:1120:38)
#49     _CustomZone.run (dart:async/zone.dart:1021:19)
#50     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
#51     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
```

LastMonopoly, 2019-04-12:
> Just guessing- but have you tried without CharacteristicWriteType.withResponse ? I found that to be a problem before on iOS.

If we don't use CharacteristicWriteType.withResponse, what alternatives do we have?

ericstephen, 2019-04-17:
> > Just guessing- but have you tried without CharacteristicWriteType.withResponse ? I found that to be a problem before on iOS.
> 
> If we don't use CharacteristicWriteType.withResponse, what alternatives do we have?

CharacteristicWriteType.withoutResponse

BradHacker, 2019-04-20:
> > > Just guessing- but have you tried without CharacteristicWriteType.withResponse ? I found that to be a problem before on iOS.
> > 
> > 
> > If we don't use CharacteristicWriteType.withResponse, what alternatives do we have?
> 
> CharacteristicWriteType.withoutResponse

Using CharacteristicWriteType.withoutResponse, the error doesn't show, but the information is never sent.

sonnny, 2019-05-22:
Brad,
    Just partially used your example and it's working for me. main.dart here -> http://sayaw.net/main_dart.txt

zwells0225, 2019-06-28:
Please note sonny's example requires version 0.5.0
Version 0.6.0 will break it

BradHacker, 2019-07-16:
For some reason, the code just recently began working using `CharacteristicWriteType.withoutResponse`. I think I updated flutter so not sure if there were any updates during that time. My code is [here](https://github.com/BradHacker/cycler-controller)

closed 2019-07-16 by BradHacker

siddharthshah87, 2019-07-22:
I am having the same issue with the example application provided with flutter_blue. Is there a particular version of fluttter_blue(0.5.0?) and a certain version of flutter the only solution to fix this issue?

xuzhengquan, 2019-07-23:
hey, guys, I am having the same issue, any update?

VaivanOmmeren, 2019-07-25:
Also getting the status 128 issue when my message size exceeds 244 bytes.

BradHacker, 2019-07-25:
My code works using `flutter_blue: ^0.5.0`. I only send a few bytes of info to keep latency low. Again, a link to my working code: https://github.com/BradHacker/cycler-controller

There is an Arduino on the other end that reads the values sent and does some logic to control a humanoid robot. Everything seems to work bluetooth wise. Having some issues on the Arduino end that have nothing to do with bluetooth (primarily being just plain out of pins using an Uno).

I'm a student and currently on summer break so I have to wait till September to begin working again on the robot. We ordered some [Teensy 3.6](https://www.sparkfun.com/products/14057) boards to replace them which will only improve the processing time. I highly recommend these boards.

kamilk91, 2019-10-16:
flutter_blue 0.6.2 ```PlatformException (PlatformException(write_characteristic_error, service (daebb240 - 00ffffffff ) could not be located on the device, null))```method used:
```
await vibChar.write(startVibrating, withoutResponse: true);
```

how flutter_blue knows that he is talking to correct device? im writing simply to characteristic, and i have no place to specify device.

jacobokoenig, 2019-12-29:
I have this issue specifically if I connect to a bottle, write and get data successfully, disconnect, but then try to connect again. Any ideas why this might be?

realrk69, 2020-11-28:
Guys, the code and troubleshooting in this thread helped me build my product faster. Parts of the code in the question and answers should be included in the official documentation and readme. I spent almost 6 hours trying to find answers online and in code.

yasasmi, 2021-04-18:
hi, I have been trying to connect mi band 5 to my flutter app using flutter_blue but I'm getting an empty array which means characteristics are null. I can't find the issue why specifically happening this and if I can get help to solve this issue it will be a great help. 

thank you for your time and considerations!