pauldemarco/flutter_blue #227: Incredibly slow/few scan results on iPhones <7 (tested on 5+6)

chu86, 2019-03-27:
Hello everyone. 
We are currently creating an app that relies on scanning for BLE advertisements. 
We have noticed that there are barely any, some none at all results in several seconds of scanning. 
Other devices consistently produce 15+ results a second for our Beacon. 
Trying to eliminate problem sources and so far my best guess is the hardware?

Have any of you run flutter_blue on older iPhones with similar or better results?

fvisticot, 2019-03-27:
Hello,
I get the same pb !!
A ticket is already opened but no news :(

[iOS] Scan method is loosing peripheral detections #211

I plan to redevelop a more simple plugin only for scanning purpose if nothing is moving on this current issue.

chu86, 2019-04-03:
I have an update on this. 
We have noticed that these problems (at least in our usecase) can be 100% eliminated when we constantly slightly update our advertisement value. iPhones <7 seem to have some kind of power saving going on when advertisement data stays static - this results in very few actual scan results appearing when they do not change. 

So our solution, since we do use this very dynamic anyway, is have the sender slightly update one of the payload values all the time.

closed 2019-04-03 by chu86

riccardoratta, 2019-04-10:
In my case a solution was to relaunch a new scan every 2/3 second, have you tried this approach?