pauldemarco/flutter_blue #230: Unhandled Exception: PlatformException(no_permissions, flutter_blue plugin requires location permissions for scanning, null)

kushtra24, 2019-04-05:
Getting an error on start

```
E/BpSurfaceComposerClient(21920): Failed to transact (-1)
E/BpSurfaceComposerClient(21920): Failed to transact (-1)
E/flutter (21920): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(no_permissions, flutter_blue plugin requires location permissions for scanning, null)
E/flutter (21920): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:564:7)
E/flutter (21920): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:302:33)
E/flutter (21920): <asynchronous suspension>
E/flutter (21920): #2      FlutterBlue.scan (file:///C:/src/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.5.0/lib/src/flutter_blue.dart:87:20)
E/flutter (21920): <asynchronous suspension>
E/flutter (21920): #3      ConnectToCarState._startAnimation (package:presentation_app/ui/connect_to_car.dart:47:10)
E/flutter (21920): #4      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:513:14)
E/flutter (21920): #5      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:568:30)
E/flutter (21920): #6      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:120:24)
E/flutter (21920): #7      TapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:242:9)
E/flutter (21920): #8      TapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:175:7)
E/flutter (21920): #9      PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:369:9)
E/flutter (21920): #10     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:73:12)
E/flutter (21920): #11     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:101:11)
E/flutter (21920): #12     _WidgetsFlutterBinding&BindingBase&GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:214:19)
E/flutter (21920): #13     _WidgetsFlutterBinding&BindingBase&GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:192:22)
E/flutter (21920): #14     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:149:7)
E/flutter (21920): #15     _WidgetsFlutterBinding&BindingBase&GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:101:7)
E/flutter (21920): #16     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:85:7)
E/flutter (21920): #17     _rootRunUnary (dart:async/zone.dart:1136:13)
E/flutter (21920): #18     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (21920): #19     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (21920): #20     _invoke1 (dart:ui/hooks.dart:223:10)
E/flutter (21920): #21     _dispatchPointerDataPacket (dart:ui/hooks.dart:144:5)
E/flutter (21920): 
```
ways to reproduse

```
class ConnectToCar extends StatefulWidget {
  @override
  ConnectToCarState createState() => new ConnectToCarState();
}

class ConnectToCarState extends State<ConnectToCar> with SingleTickerProviderStateMixin {

  var scanSubscription;
  StreamSubscription _scanSubscription;

  FlutterBlue flutterBlue = FlutterBlue.instance;

  Map<DeviceIdentifier, ScanResult> scanResults = new Map();

  bool isScanning = false;

  AnimationController _controller;

  @override
  void initState() {
    super.initState();
    _controller = new AnimationController(
      vsync: this,
    );
    //_startAnimation();
  }

  @override
  void dispose() {
    _controller.dispose();
    scanSubscription.cancel();
    super.dispose();
  }

  void _startAnimation() {

    _scanSubscription = flutterBlue
        .scan(
      timeout: const Duration(seconds: 5),
      /*withServices: [
          new Guid('0000180F-0000-1000-8000-00805F9B34FB')
        ]*/
    )
        .listen((scanResult) {

//      print('localName: ${scanResult.advertisementData.localName}');
//      print(
//          'manufacturerData: ${scanResult.advertisementData.manufacturerData}');
//      print('serviceData: ${scanResult.advertisementData.serviceData}');
//      print("device type: ${scanResult.device.type.toString()}");
//      print("device type: ${scanResult.device.id.id}");

      setState(() {
        scanResults[scanResult.device.id] = scanResult;
      });

    }, onDone: _stopScan);

    setState(() {
      isScanning = true;
    });

    _controller.stop();
    _controller.reset();
    _controller.repeat(
      period: Duration(seconds: 1),
    );
  }

  _stopScan() {
    _scanSubscription?.cancel();
    _scanSubscription = null;
    setState(() {
      isScanning = false;
    });
  }

  @override
  Widget build(BuildContext context) {
    return new Scaffold(
      body: Column(
        children: <Widget>[
          pageTitle(),
          vehicle(),

          Expanded(
            child: ListView.builder(
                itemBuilder: (BuildContext context,int index) {
                  return
                    ListTile(
                    title: Text('Data: $scanResults'),
                  );
                }
            ),
          ),
        ],
      ),

      floatingActionButton: new FloatingActionButton(
        onPressed: _startAnimation,
        child: new Icon(Icons.play_arrow),
      ),
    );
  }
```

ened, 2019-04-06:
@kushtra24 the message is clear, you need to fetch permissions (location) first, before you can scan for nearby devices.

cingh-jasdeep, 2019-06-24:
> @kushtra24 the message is clear, you need to fetch permissions (location) first, before you can scan for nearby devices.
It doesn't ask for permissions even if I add to the manifest

closed 2020-04-02 by pauldemarco