pauldemarco/flutter_blue #231: Android Connection Status 133 - Cannot Connect

bretmh, 2019-04-05:
I've been having this issue with specifically android devices. iOS works perfectly fine and i've tried to change {autoConnect} to false as others have tried.

I've been testing with a MotoG3 and MotoX.

Here is my code for the connection and scan. Note this code works perfectly fine for iOS..

**Start Scan**
` 
  _startScan() {
    _scanSubscription = _flutterBlue
        .scan(
      timeout: const Duration(seconds: 5),
    )
        .listen((scanResult) {
      if (scanResult.advertisementData.localName.toString() == "mydevice") {
        print('Device Name: ${scanResult.advertisementData.localName}');
        print('Connectable: ${scanResult.advertisementData.connectable}');
        print('Service UUIDs: ${scanResult.advertisementData.serviceUuids}');
        setState(() {
          scanResults[scanResult.device.id] = scanResult;
        });
      }
    }, onDone: _stopScan);

    setState(() {
      isScanning = true;
    });
  }
`

**Connect**
`
  _connect(BluetoothDevice d) async {
    print("Trying to connect....");

    device = d;

    deviceConnection =
        _flutterBlue.connect(device, autoConnect: false).listen((s) {
      if (s == BluetoothDeviceState.connected) {
        print("Connected");

        setState(() {
          //since we're connected update state
          deviceState = s;

          //Subscribe to Device State Changes.
          deviceStateSubscription = device.onStateChanged().listen((s) {
            setState(() {
              deviceState = s;
            });
          });
          print("Subscribed to Connection State Changes.");

          //Discover device services.
          device.discoverServices().then((s) {
            setState(() {
              print("Services discovered");
              services = s;
            });
          });
        });
      } else {
        print("Not Connected");
      }
    });
  }
`

jthansen, 2019-04-12:
I am having the same issue. It works perfectly on iOS and then I am unable to get scan results on android using Pixel XL.

jthansen, 2019-04-12:
My issue was that even though location was enabled in the specific App permissions, it was not "on" under Security & location.

bretmh, 2019-04-12:
> My issue was that even though location was enabled in the specific App permissions, it was not "on" under Security & location. Now it's working perfectly.

Could you possibly share your code? I'm particularly having an issue on connection. Haven't had the time to mess with it. I feel like I'm missing something obvious.

jthansen, 2019-04-12:
Your connect method works fine with the [Flutter Blue example](https://github.com/pauldemarco/flutter_blue/blob/master/example/lib/main.dart) so I'm not sure what's going on. And your scanning works fine? You're able to find your device?  Do you need toString() with localName?

bretmh, 2019-04-12:
> Your connect method works fine with the [Flutter Blue example](https://github.com/pauldemarco/flutter_blue/blob/master/example/lib/main.dart) so I'm not sure what's going on. And your scanning works fine? You're able to find your device? Do you need toString() with localName?

I can find my devices on both android and ios. But it's my connecting that I get an issue with. Yes you are right I don't need toString.

bretmh, 2019-04-12:
Updated my primary post with my error code. I thought I had it up there.

```
I/flutter ( 7378): Trying to connect....
D/BluetoothManager( 7378): getConnectedDevices
D/BluetoothGatt( 7378): connect() - device: 0C:2A:69:11:33:30, auto: false
D/BluetoothGatt( 7378): registerApp()
D/BluetoothGatt( 7378): registerApp() - UUID=665b1de4-43d0-4153-9152-6612bfc0291a
D/BluetoothGatt( 7378): onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt( 7378): onClientConnectionState() - status=133 clientIf=5 device=0C:2A:69:11:33:30
D/FlutterBluePlugin( 7378): [onConnectionStateChange] status: 133 newState: 0
I/flutter ( 7378): Not Connected
```

lcrocker, 2019-04-17:
I fixed my status 133 errors after reading:

https://stackoverflow.com/questions/25330938/android-bluetoothgatt-status-133-register-callback

FlutterBluePlugin.java, line 214, change:

BluetoothGatt gattServer = device.connectGatt(registrar.activity(), options.getAndroidAutoConnect(), mGattCallback);

to

BluetoothGatt gattServer = device.connectGatt(registrar.activity(), options.getAndroidAutoConnect(), mGattCallback, BluetoothDevice.TRANSPORT_LE);

mainini, 2019-04-26:
PR #229 fixes this...

pauldemarco, 2019-07-15:
I can confirm PR #229 did resolve this issue with an LG G6.

Please re-open if you are experiencing this after trying the fix, and supply the phone model you are using.

closed 2019-07-15 by pauldemarco