pauldemarco/flutter_blue #234: Can't check if Bluetooth is available

remygautron, 2019-04-19:
Hi, I have a problem.
I can't check if Bluetooth is available or not, my code :
```dart
FlutterBlue.instance.isAvailable.then((value) => {
      text = "Compatible"
    }).catchError((e) => {
      text = "Incompatible"
    }, test: (e) => e is PlatformException);
```

I always have the error in the console, and my var still null.

riccardoratta, 2019-04-23:
Can you add the error that you get in the console?

riccardoratta, 2019-04-23:
This code should work to check when the bluetooth is available; could you try this and see if you get any error?

```dart
FlutterBlue.instance.isAvailable.then((bool isAvailable) {
  if (isAvailable) {
    print('bluetooth is available');
  } else {
    print('bluetooth is NOT available, device may be incompatible');
  }
});
```

remygautron, 2019-04-23:
> This code should work to check when the bluetooth is available; could you try this and see if you get any error?
>
> ```dart
> FlutterBlue.instance.isAvailable.then((bool isAvailable) {
>   if [...]

The code work thanks !

closed 2019-04-23 by remygautron