pauldemarco/flutter_blue #238: Doesn't find any devices until I turn on location service

mehdimsv, 2019-04-23:
first time I scaned I didn't get any devices on the list but after turning location service on,I got some devices shown by their manufacture ids, not by UUID.
please tell me what this has to do with location service, I can't request for location service to be turned on from my customer.
thanks

Jogosb, 2019-04-25:
Hi! I'm just doing some flutter research on BLE and bumped into this issue. It's not issue with the library but with the Android itself. 
Please check this issue report https://issuetracker.google.com/issues/37060483 
Basically to perform scanning on post Android 6.0 devices locations services have to be turned on. 
And also you actually can open a pop up for a user to turn them on! Hope it helps, cheers!

riccardoratta, 2019-04-25:
Android require to ask the location permission for performing a bluetooth scan. This is because you can, in theory, deduct the position of your user looking at the scan result (using the MAC unique address and the rssi value).

mehdimsv, 2019-04-25:
thanks.
I can close the issue now.

closed 2019-04-25 by mehdimsv