pauldemarco/flutter_blue #244: Unable to build for IOS for weeks

LiveRock, 2019-04-27:
I have been trying and trying without success for weeks...
Cannot build for ios. Android ok.
here are the errors:
Xcode build done.                                           125.1s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **
Xcode's output:
↳
    /ios/Pods/Bolts/Bolts/iOS/BFAppLinkReturnToRefererController.m:133:17: warning: block implicitly retains 'self'; explicitly mention 'self' to indicate this is intended behavior [-Wimplicit-retain-self]
                    _view.frame = CGRectMake(0.0, 0.0, CGRectGetWidth(_view.bounds), 0.0);
                    ^
                    self->
   /ios/Pods/Bolts/Bolts/iOS/BFAppLinkReturnToRefererController.m:133:67: warning: block implicitly retains 'self'; explicitly mention 'self' to indicate this is intended behavior [-Wimplicit-retain-self]
                    _view.frame = CGRectMake(0.0, 0.0, CGRectGetWidth(_view.bounds), 0.0);
                                                                      ^
                                                                      self->
   /ios/Pods/Bolts/Bolts/iOS/BFAppLinkReturnToRefererController.m:148:18: warning: block implicitly retains 'self'; explicitly mention 'self' to indicate this is intended behavior [-Wimplicit-retain-self]
                    [_view sizeToFit];
                     ^
                     self->
    ios/Pods/Bolts/Bolts/iOS/BFAppLinkReturnToRefererController.m:189:13: warning: block implicitly retains 'self'; explicitly mention 'self' to indicate this is intended behavior [-Wimplicit-retain-self]
/ios/Pods/Protobuf/objectivec/google/protobuf/Struct.pbobjc.m:19:10: warning: non-portable path to file '<protobuf/Struct.pbobjc.h>'; specified path differs in case from file name on disk [-Wnonportable-include-path]
     #import <Protobuf/Struct.pbobjc.h>
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
             <protobuf/Struct.pbobjc.h>
    1 warning generated.
    ios/Pods/Protobuf/objectivec/google/protobuf/Timestamp.pbobjc.m:17:10: warning: non-portable path to file '<protobuf/Timestamp.pbobjc.h>'; specified path differs in case from file name on disk [-Wnonportable-include-path]
     #import <Protobuf/Timestamp.pbobjc.h>
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             <protobuf/Timestamp.pbobjc.h>
    1 warning generated.
    /ios/Pods/Protobuf/objectivec/google/protobuf/Type.pbobjc.m:19:10: warning: non-portable path to file '<protobuf/Type.pbobjc.h>'; specified path differs in case from file name on disk [-Wnonportable-include-path]
     #import <Protobuf/Type.pbobjc.h>

This is just s snapshot. It is a very long error list.

Any idea what went wrong?

lmarceau, 2019-05-01:
Hi there, got a couple questions for you:
1. what does your `flutter doctor` says?
2. you're showing some clang warnings, can you show us the error(s) that make the failure?
3. is this error caused by flutter_blue?

bjornwilliam, 2019-05-02:
There is a known pod issue, mentioned in a few issues.  Look at https://github.com/pauldemarco/flutter_blue/issues/108 too see if this is the case for you. 

My issue at least was in flutter_blue.podspec, where I had to change protoc = ENV['PWD'] + '/Pods/!ProtoCompiler/protoc'  to  protoc = "#(PODS_ROOT)/!ProtoCompiler/protoc"

NOTE the parantheses around PODS_ROOT. Try either {} or (). I had to use parantheses.

lmarceau, 2019-05-02:
If this is the cause of this issue, then another temporary fix if you don't want to modify the podspec is to `ln -s ios/Pods` into your project root folder. This will create a symlink so your Pods can be found.

LiveRock, 2019-05-03:
I read about all the above before and I tried everything. Finally, I did a lot of other stuffs that I read online and I believe it is a combination of
1. Pod deintegrate
2. Pod upadte
3. added config.build_settings['SWIFT_VERSION'] = '4.3' to podfile
4. Copy a missing runner.entitlement from another project

And finally, it build and ran. Thanks

closed 2019-06-27 by LiveRock