pauldemarco/flutter_blue #252: Connnect BLE devcie error,  under flutter v1.5.9 pre147

guixue, 2019-05-08:
I run the example under latest flutter (1.5.9 pre147) version,  met an error.

SCAN action is ok,   the problem occurred while click CONNECT button.

detail error message as below:

```
D/BluetoothGatt( 4136): onClientConnectionState() - status=0 clientIf=5 device=10:CE:A9:2F:73:3F
D/FlutterBluePlugin( 4136): [onConnectionStateChange] status: 0 newState: 2
W/BluetoothGatt( 4136): Unhandled exception in callback
W/BluetoothGatt( 4136): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: Binder:4136_3
W/BluetoothGatt( 4136): 	at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:605)
W/BluetoothGatt( 4136): 	at io.flutter.embedding.engine.FlutterJNI.dispatchPlatformMessage(FlutterJNI.java:515)
W/BluetoothGatt( 4136): 	at io.flutter.embedding.engine.dart.DartMessenger.send(DartMessenger.java:74)
W/BluetoothGatt( 4136): 	at io.flutter.embedding.engine.dart.DartExecutor.send(DartExecutor.java:163)
W/BluetoothGatt( 4136): 	at io.flutter.view.FlutterNativeView.send(FlutterNativeView.java:152)
W/BluetoothGatt( 4136): 	at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:95)
W/BluetoothGatt( 4136): 	at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:82)
W/BluetoothGatt( 4136): 	at com.pauldemarco.flutterblue.FlutterBluePlugin$8.onConnectionStateChange(FlutterBluePlugin.java:757)
W/BluetoothGatt( 4136): 	at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:262)
W/BluetoothGatt( 4136): 	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
W/BluetoothGatt( 4136): 	at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
W/BluetoothGatt( 4136): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:257)
W/BluetoothGatt( 4136): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
W/BluetoothGatt( 4136): 	at android.os.Binder.execTransact(Binder.java:735)
D/BluetoothManager( 4136): getConnectionState()
D/BluetoothManager( 4136): getConnectedDevices
I/flutter ( 4136): State refreshed: BluetoothDeviceState.connected
D/BluetoothGatt( 4136): onConnectionUpdated() - Device=10:CE:A9:2F:73:3F interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 4136): onConnectionUpdated() - Device=10:CE:A9:2F:73:3F interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt( 4136): close()
D/BluetoothGatt( 4136): unregisterApp() - mClientIf=5
```

flutter_blue: 5.0(git: 6f6fa30d77c7d935)

Any one can give some advice, thank you!

mcrovero, 2019-05-08:
I'm having the same issue

mopilo, 2019-05-11:
any fix yet?

mopilo, 2019-05-21:
any solution to this yet please?

sonnny, 2019-05-22:
I downgrade flutter to v1.3.0 and it works, code here ------> http:// sayaw.net/main_dart.txt

guixue, 2019-05-28:
the warning maybe coursed by this flutter engine patch: https://github.com/flutter/engine/commit/2c9e37c34e79475bbde7c8163eb5e56cdb9662a1

guixue, 2019-05-28:
I refer to this solution:
https://github.com/pichillilorenzo/flutter_inappbrowser/issues/98

erik444, 2019-06-03:
> 
> 
> I downgrade flutter to v1.3.0 and it works, code here ------> http://sayaw.net/main_dart.txt

I also have the same problem and get the same message as @guixue.

Unfortunately, your solution did not work for me. Do you or anyone else have another suggestions to solve this issue?

Thanks!

mopilo, 2019-06-03:
Well, I had to use this https://pub.dev/packages/blue_thermal_printer. works for only android

aaron3481, 2019-06-06:
Because now flutter will check if the data passing out from MainThread

A-Dirt, 2019-06-12:
@pauldemarco I have the same problem under the lastest version(1.7.3), I look forward to your solving this problem as soon as possible!

dmvvilela, 2019-06-18:
Same issue here, will they fix this in this plugin?

casiff, 2019-07-12:
Is there now a solution for using flutter 1.7.8 with flutter_blue: 0.5 ?
At the moment I am going back to flutter 1.2.1.

bretie, 2019-07-18:
All of a sudden I started experiencing this... Is there a fix or do we just downgrade flutter?

casiff, 2019-07-19:
I am using flutter 1.7.8 with the fixed flutter_blue 0.5 version of https://github.com/bpillon/flutter_blue.

earyzhe, 2019-08-22:
I'm also experiencing this error:

[✓] Flutter (Channel stable, v1.7.8+hotfix.4, on Mac OS X 10.14.6 18G84, locale en-GB)
[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 10.3)
[✓] iOS tools - develop for iOS devices
[✓] Android Studio (version 3.5)
[✓] IntelliJ IDEA Community Edition (version 2018.3.4)
[✓] VS Code (version 1.37.1)

pauldemarco, 2019-09-11:
This should be resolved in the version as of June 19.
If not, please re-open the issue.

closed 2019-09-11 by pauldemarco