pauldemarco/flutter_blue #254: Can't build when AndroidX

remygautron, 2019-05-14:
Hello,

When the Android project is migrated to AndroidX, the build doesn't work.

How to reproduce :

1) Add geolocator or location plugin for example..
2) Migrate the Android project to AndroidX
3) Add flutter_blue
4) Build the project

smccleary, 2019-05-15:
Same here. Looking at the back and forth on similar issues in this repo, I take it this library does not work on this newer version of android.

FWeissenb, 2019-05-29:
@remygautron Do you have any build logs?
We are using this plugin with androidx and it works when changing protobuf-gradle-plugin version from 0.8.7 to 0.8.8.

Its in the build.gradle in .pub-cache 

Look here https://github.com/pauldemarco/flutter_blue/pull/253

pauldemarco, 2019-06-19:
Please try the latest version of the plugin and re-open if the issue persists.

closed 2019-06-19 by pauldemarco