pauldemarco/flutter_blue #255: Flutter run on ios, pod install error

paranoiasystem, 2019-05-14:
Hi, I'm trying to launch the command "flutter run" on an iOS device, but when I try it, I receive the following error:

```
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/installer/xcode/target_validator.rb:115:in `verify_swift_pods_swift_version'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/installer/xcode/target_validator.rb:37:in `validate!'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/installer.rb:459:in `validate_targets'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/installer.rb:138:in `install!'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/command/install.rb:48:in `run'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/claide-1.0.2/lib/claide/command.rb:334:in `run'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/lib/cocoapods/command.rb:52:in `run'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/gems/cocoapods-1.6.1/bin/pod:55:in `<top (required)>'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/bin/pod:22:in `load'
    /usr/local/Cellar/cocoapods/1.6.1/libexec/bin/pod:22:in `<main>'

Error output from CocoaPods:
↳

       protoc:0: warning: File 'flutterblue.proto' has no package. Consider adding a new package to the proto and adding 'new.package = Protos' to the expected prefixes
       file ().

Error running pod install
```

I have already tried some solutions found in the issues section, but it doesn't work for me, can someone help me?

ndrd, 2019-05-14:
same issue here! 

[!] `<PBXGroup UUID=`97C146E51CF9000F007C117D`>` attempted to initialize an object with an unknown UUID. `CF3B75C9A7D2FA2A4C99F110` for attribute: `children`. This can be the result of a merge and  the unknown UUID is being discarded.

pauldemarco, 2019-06-19:
Please try the latest version of the plugin and re-open if the issue persists.

closed 2019-06-19 by pauldemarco

tanliubin, 2019-08-12:
Have you solved it? @paranoiasystem