pauldemarco/flutter_blue #257: Update plugin to latest flutter structure, and protobuf

pauldemarco, 2019-05-15:
This plugin needs a face-lift to the latest template from flutter, and an overall dependency update (protobuf)

Related to the following:
Issue #255 
Issue #254 
Issue #247 
Issue #212 
Issue #206 
Issue #209 
Issue #200

FWeissenb, 2019-05-20:
Any timeline for this? Or a timeline integrating all the PRs?

nightscape, 2019-05-30:
Hey @pauldemarco, I hope I'm not stepping on your toes, but this project is starting to show an unhealthy pattern of issues and PRs piling up. I'm blindly guessing that you didn't expect the interest in this project to be so huge and are probably maintaining it in your spare time.
One option you might consider is picking one or a few PRs authors that did a good job and fulfilled your code and communication standards, and granting them commit rights to this repo to reduce the bottleneck of your probably very limited time.
I've seen a lot of great projects die because at one place they got into a state where every user needed to find a fork that fixes a certain bug they were facing, then a few people start cherry-picking commits from PRs in their repos and in the end it becomes such a complicated mess that noone wants to touch that project anymore...

pauldemarco, 2019-05-30:
@nightscape Not at all.  I am also heavily invested in this plugin with products both on the market and under development.  They are getting to a point where I can focus more time on the software, and flutter_blue will get most of that attention. @FWeissenb I am expecting this to take place end of June.

All I can ask is for flutter_blue users to go through issues they are facing and leave a 👍 . This allows me to effectively triage and get going as quickly as possible.

Thanks again everyone for your interest in this project!

ygxdha, 2019-06-02:
i've already update my flutter to v1.6.3 and protobuf to ^0.10.5
if i use `dependencies {
        classpath 'com.android.tools.build:gradle:3.2.1'
    }` in build.gradle
it's work fine.and if i change to `dependencies {
        classpath 'com.android.tools.build:gradle:3.4.1'
    }`

build app still report 

> The com.google.protobuf plugin was already applied to the project: :bluetooth and will not be applied again after plugin:

FWeissenb, 2019-06-07:
> @nightscape Not at all. I am also heavily invested in this plugin with products both on the market and under development. They are getting to a point where I can focus more time on the software, and [...]

I think it's still a good idea from @nightscape to add more person that can handle pull requests. 
It's fine if you have time to maintain this project end of June. But what happens in the future? Flutter will make some updates that break the plugin and we have to fork it and fix the bug that our apps will work and wait that you will merge the bugfix 3 months later. 
I can understand that it's "your baby" but share the responsibility would be a nice idea. So we don't get on your nerves every day :) 
Think about it.

audkar, 2019-06-07:
Maybe https://github.com/fluttercommunity could be possible option?

audkar, 2019-06-11:
Thank you for maintaining this plugin. 
Just tip that - seems that this project was copied https://pub.dev/packages/flutter_ble . That can't be called hard fork, because authors removed Licence :P

pauldemarco, 2019-06-19:
Everyone subscribed to this issue, please implement the latest version of the plugin and let me hear your thoughts.

Thanks!

closed 2019-06-19 by pauldemarco