pauldemarco/flutter_blue #261: not found android phone

azaz0133, 2019-05-23:
i'm used example flutter_blue for test and i scan. but not found devices thai is phone such as another phone that turn ou blutetooth already.

ycktw, 2019-05-24:
Hi, It seems to be also mentioned in #238 --
You can turn on location service (GPS) first.
And have another try.

azaz0133, 2019-05-24:
> Hi, It seems to be also mentioned in #238 --
> You can turn on location service (GPS) first.
> And have another try.

i followed uyour suggestion. but it not worked when i scaned i found all devices without ipad iphone and android phone at all. Devices that i found is Mi band3 and any accessories

pauldemarco, 2019-06-18:
@azaz0133 What type of device are you looking for? Is it a Bluetooth Low Energy device?

pauldemarco, 2020-04-02:
Stale.

closed 2020-04-02 by pauldemarco