pauldemarco/flutter_blue #264: Plugin always causes exception when Android device doesn't support bluetooth

hikaritenchi, 2019-05-28:
The constructor for `FlutterBlue flutterBlue = FlutterBlue.instance;` calls `setLogLevel(logLevel);` which is a platform call.

That call will throw an exception if the `mBluetoothAdapter` is null, and if we're calling anything other than `isAvailable()`. Since the device doesn't have bluetooth, `mBluetoothAdapter` is null, and since we're calling `setLogLevel()`, it just throws an exception.

There's no way to actually call `isAvailable`, because the instance constructor's call precedes any attempt to use `isAvailable`.

I think the solution is to add `setLogLevel` to the `if` statement at the top of `onMethodCall`.

benkuper, 2019-12-22:
+1 this renders the .isAvailable() function useless.
Another way of fixing it is to put the isAvailable function as static (and not using channel but instancing a temporary methodClass) :
``` 
static Future<bool> get isAvailable =>
    MethodChannel('$NAMESPACE/methods').invokeMethod('isAvailable').then<bool>((d) => d);
```

This way, you can do `Flutterblue.isAvailable.then( (value) { ... });`

benkuper, 2019-12-22:
+1 this renders the .isAvailable() function useless, and problematic when working on virtual devices.
Another way of fixing it is to put the isAvailable function as static (and not using channel but instancing a temporary methodClass) :

static Future<bool> get isAvailable =>
    MethodChannel('$NAMESPACE/methods').invokeMethod('isAvailable').then<bool>((d) => d);
This way, you can do Flutterblue.isAvailable.then( (value) { ... });

Tested and working.

ride4sun, 2020-03-23:
@pauldemarco  Am I wrong - would that not result in an Exception for any Example of this plugin ut of the box?
I just did run into this problem. Is there a fix. Is this not a critical defect?

benkuper, 2020-03-24:
As I am not at all an expert in flutter (3 days of working with it total), I have no clue if what I'm proposing is good practice so I haven't put a pull request for that, no.

MbarkErras, 2020-04-06:
@pauldemarco is there any solution to this, I tried everything I found on the internet!