pauldemarco/flutter_blue #265: How Do I Send Data to Other Phones?

patientplatypus, 2019-05-29:
This is very frustrating. I have examples of how to read from other devices, but I don't have any examples on how to publish data. So I have an app where I want to change my advertisement data to say "This is this app thing! You should connect to me!", which would be useful. I also don't have any indication from the documentation on how I'm actually supposed to publish a service for other applications to consume. There's lots of request information here, but nothing on how to serve data. How am I supposed to use this?

MasiaVittorio, 2019-07-07:
Exactly this. I just want to find a way to make N identical apps on N different phones able to communicate with each other via BLE, but it seems impossible.

Mooibroek, 2019-07-22:
Im looking for exactly this for an app im building for a client. Im willing to add the functionality myself, but lack some Bluetooth knowledge on how to get started. Can anybody point me in the right direction on how to do this natively on iOS and Android?

Scowley4, 2021-09-09:
@patientplatypus @MasiaVittorio @Mooibroek Have any of you made any progress on this? Basically I'm hoping to play with "N identical apps on N different phones able to communicate with each other".