pauldemarco/flutter_blue #267: Include a 'getBondedDevices()' API

steenson, 2019-06-06:
It would be very useful to have an API which returns a list of bonded devices. These devices are saved by the phone when a pairing/bonding process has happened between the phone and the peripheral.

This is implemented in the 'flutter_blue_serial' plug-in which uses flutter blue as a base.
It's implemented like this:

```
Future<List<BluetoothDevice>> getBondedDevices() async {
    final List list = await _channel.invokeMethod('getBondedDevices');
    return list.map((map) => BluetoothDevice.fromMap(map)).toList();
  }
```

And  then:

```
case "getBondedDevices":
                try {

                    if (ContextCompat.checkSelfPermission(registrar.activity(),
                            Manifest.permission.ACCESS_COARSE_LOCATION)
                            != PackageManager.PERMISSION_GRANTED) {

                        ActivityCompat.requestPermissions(registrar.activity(),
                                new String[]{Manifest.permission.ACCESS_COARSE_LOCATION},
                                REQUEST_COARSE_LOCATION_PERMISSIONS);

                        pendingResult = result;
                        break;
                    }

                    getBondedDevices(result);

                } catch (Exception ex) {
                    result.error("Error", ex.getMessage(), exceptionToString(ex));
                }

                break;
```

and then:

```
/**
     * @param result result
     */
    private void getBondedDevices(Result result) {

        List<Map<String, Object>> list = new ArrayList<>();

        for (BluetoothDevice device : mBluetoothAdapter.getBondedDevices()) {
            Map<String, Object> ret = new HashMap<>();
            ret.put("address", device.getAddress());
            ret.put("name", device.getName());
            ret.put("type", device.getType());
            list.add(ret);
        }

        result.success(list);
    }
```

ericstephen, 2019-09-17:
This should really be a thing. Please bring this to the platform.

spsarolkar, 2020-04-07:
Anyone working on this?