pauldemarco/flutter_blue #269: Connect glucometer flutter

david-gam, 2019-06-08:
I'm trying to connect a meter with flutter but I could not write or receive the stored readings or the new glucose readings, can anyone have an example or know how I can solve this?
glucose service: 00001808-0000-1000-8000-00805f9b34fb
glucose measurement: 00002a18-0000-1000-8000-00805f9b34fb
RECORDS CHARACTERISTIC: 00002a52-0000-1000-8000-00805f9b34fb
CLIENT CHARACTERISTIC CONFIG: 00002902-0000-1000-8000-00805f9b34fb

FWeissenb, 2019-06-11:
What's the specific problem? Do you have any error message? Or some code how do you try to read the values?

pauldemarco, 2019-06-18:
This question is better suited for a QA site, like [stackoverflow](https://stackoverflow.com), where it can be indexed for others to learn from as well.

Please provide code snippets and any other information that might lead you to believe it is a bug with the plugin, and we can re-open the issue.

closed 2019-06-18 by pauldemarco