pauldemarco/flutter_blue #270: java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: Binder_1

cfh763551832, 2019-06-10:
D/BluetoothAdapter(23338): STATE_ON
D/BluetoothLeScanner(23338): onClientRegistered() - status=0 clientIf=7
D/BluetoothAdapter(23338): STATE_ON
D/BluetoothManager(23338): getConnectedDevices
D/BluetoothGatt(23338): connect() - device: 88:1B:99:08:18:AF, auto: true
D/BluetoothGatt(23338): registerApp()
D/BluetoothGatt(23338): registerApp() - UUID=3440121c-5ac6-47c4-89e6-cc0f7314c8a8
D/BluetoothGatt(23338): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(23338): onClientConnectionState() - status=133 clientIf=7 device=88:1B:99:08:18:AF
D/FlutterBluePlugin(23338): [onConnectionStateChange] status: 133 newState: 0
W/BluetoothGatt(23338): Unhandled exception in callback
W/BluetoothGatt(23338): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: Binder_1
W/BluetoothGatt(23338): 	at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:605)
W/BluetoothGatt(23338): 	at io.flutter.embedding.engine.FlutterJNI.dispatchPlatformMessage(FlutterJNI.java:515)
W/BluetoothGatt(23338): 	at io.flutter.embedding.engine.dart.DartMessenger.send(DartMessenger.java:76)
W/BluetoothGatt(23338): 	at io.flutter.embedding.engine.dart.DartExecutor.send(DartExecutor.java:166)
W/BluetoothGatt(23338): 	at io.flutter.view.FlutterNativeView.send(FlutterNativeView.java:155)
W/BluetoothGatt(23338): 	at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:98)
W/BluetoothGatt(23338): 	at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:84)
W/BluetoothGatt(23338): 	at com.pauldemarco.flutterblue.FlutterBluePlugin$8.onConnectionStateChange(FlutterBluePlugin.java:757)
W/BluetoothGatt(23338): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181)
W/BluetoothGatt(23338): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
W/BluetoothGatt(23338): 	at android.os.Binder.execTransact(Binder.java:453)

cfh763551832, 2019-06-11:
这个问题是flutter版本问题，我用flutter beta 1.5.4之后就不报错了，但是蓝牙连接不上是什么原因?

closed 2019-06-11 by cfh763551832

A-Dirt, 2019-06-11:
你报错的时候  用的什么版本的

cfh763551832, 2019-06-11:
1.6.3 beta

A-Dirt, 2019-06-11:
我用的1.7.3    也是这问题...

cfh763551832, 2019-06-11:
我改成1.54之后，就不报这个错，但是我的蓝牙一直是连不上的，一直是disconnected

yiguozhen0510, 2019-11-18:
我也遇到同样的问题了，请问你们是怎么样解决的？