pauldemarco/flutter_blue #271: BluetoothDeviceState.disconnected

cfh763551832, 2019-06-11:
D/BluetoothAdapter(16394): startLeScan(): null
D/BluetoothAdapter(16394): STATE_ON
D/BluetoothLeScanner(16394): onClientRegistered() - status=0 clientIf=7
D/BluetoothAdapter(16394): stopLeScan()
D/BluetoothAdapter(16394): STATE_ON
I/flutter (16394): devicename==QC3834
D/BluetoothManager(16394): getConnectedDevices
D/BluetoothGatt(16394): connect() - device: 88:1B:99:08:18:AF, auto: false
D/BluetoothGatt(16394): registerApp()
D/BluetoothGatt(16394): registerApp() - UUID=e76018b8-857f-43c8-9d9c-0a88216f3c24
D/BluetoothGatt(16394): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(16394): onClientConnectionState() - status=133 clientIf=7 device=88:1B:99:08:18:AF
D/FlutterBluePlugin(16394): onConnectionStateChange: 
I/flutter (16394): BluetoothDeviceState.disconnected
I/flutter (16394): 蓝牙连接失败BluetoothDeviceState.disconnected

Fleximex, 2019-06-12:
If FlutterBlue fails to connect in the time set (in the example this is 4 seconds) to the device (in your case the device with mac address 88:1B:99:08:18:AF), then it will go to the disconnected state.

pauldemarco, 2019-06-18:
@cfh763551832 As @Fleximex has stated, there is a configurable connection timeout. If your problem persists, please provide more details like a code snippet and re-open the issue.

closed 2019-06-18 by pauldemarco