pauldemarco/flutter_blue #272: IOS bluetooth request popup when start app

catacs, 2019-06-14:
When launching the app with flutter blue in IOS it pops up the dialog to activate bluetooth in the device. It shouldn't ask for that until the bluetooth is needed. Bluetooth is optional in our app and it's not user friendly to ask it each time the user starts the app.

FWeissenb, 2019-06-14:
How do you create your flutter blue instance?
Try to first create the singleton if its used.

catacs, 2019-06-14:
I am creating the singleton when it's used. You can reproduce the issue adding the flutter_blue library in the `pubspec.yaml` and starting the app with the bluetooth deactivated in the IOS device. It's only an IOS problem. The problem comes because the `centralManager` is created when in the `registerWithRegistrar` method from `FlutterBluePlugin`:
```
...
 @interface FlutterBluePlugin ()
 + (void)registerWithRegistrar:(NSObject<FlutterPluginRegistrar>*)registrar {
  ...
  instance.centralManager = [[CBCentralManager alloc] initWithDelegate:instance queue:nil];
  ...
}
 ...
@end
```

gumbypp, 2019-06-14:
You can disable it using the `CBCentralManagerOptionShowPowerAlertKey` option. See https://stackoverflow.com/questions/12533269/ios-check-if-bluetooth-is-on-without-system-alert-popup-to-user. If you create a PR, I suggest that you make it configurable - some users might want this alert.

LongXiangGuo, 2020-11-11:
I have fix it like this but it still has system alert pop up for bluetooth permission request
```Objective-C
instance.centralManager = [[CBCentralManager alloc] initWithDelegate:instance queue:nil options:@{CBCentralManagerOptionShowPowerAlertKey:@NO}];
```

LongXiangGuo, 2020-11-11:
Test with other solution  also not work for me, the system bluetooth dialog still show when the centralManager first intiializated.
 
```Objective-C
    NSDictionary *options = [NSDictionary dictionaryWithObjectsAndKeys:[NSNumber numberWithBool:NO], CBCentralManagerOptionShowPowerAlertKey, nil];
  instance.centralManager = [[CBCentralManager alloc] initWithDelegate:instance queue:nil options:options];
  [instance.centralManager scanForPeripheralsWithServices:nil options:nil];
    [instance.centralManager scanForPeripheralsWithServices:nil options:nil];
```

LongXiangGuo, 2020-11-11:
Does any one had the same problem?

LongXiangGuo, 2020-11-11:
Test with iOS 14; iPhone XS; Version 11.7 (11E801a);

ihorkozar, 2022-04-05:
@LongXiangGuo Did you manage to solve this problem?

flutter-painter, 2023-12-05:
Not specific to flutter_blue, seems like a flutter / iOS issue,
Users should cope with having bluetooth on, but indeed not pretty.
There is also a strange medium article about this issue... https://medium.com/@avengertony101/flutter-plugin-asks-for-bluetooth-permission-on-app-startup-ios-418c70693af