pauldemarco/flutter_blue #273: Reconnect to a device with Identifier

fvisticot, 2019-06-15:
On iOS, to reconnect a know peripheral with identifier:
We can get a peripheral by id with the method: retrievePeripheralsWithIdentifiers
Once we get the peripheral we can connect to it.

This method allow to not scan before connecting to device

How is it possible with Flutter_blue ?

fvisticot, 2019-06-26:
I'm still blocked on this issue. Can you please help.
I think that anybody working with a Ble device in connected mode has the same issue.

My use case:

- Device was previously connected (I know the device and his ID)
- When opening the app later, the device is not in range 
- BUT I need to re-connect the device as soon he will be in range.

Pb:
I do not see any feature in the library allowing to retrieve a know device and connect to it.

On native iOS side i will:
- retrievePeripheralsWithIdentifiers
- connect to this peripheral with a empty timeout to connect as soon as device is in range

fvisticot, 2019-06-30:
Any update on that ? would be very usefull for any application requiring to auto reconnect on application startup when device is not in range

assumptionsoup, 2019-07-05:
Unless this is an iOS only issue, I was able to do this on android:

``` dart
  BluetoothDevice device;
  void _loadState() async {
    final prefs = await SharedPreferences.getInstance();
    String deviceData = prefs.getString('device');
    if (deviceData != null) {
      device = BluetoothDevice.fromProto(
        protos.BluetoothDevice.fromJson(deviceData)
      );
    }
  }

  void _saveState() async {
    final prefs = await SharedPreferences.getInstance();
    if (device != null) {
      protos.BluetoothDevice proto = protos.BluetoothDevice.create()
        ..name = device.name
        ..type = protos.BluetoothDevice_Type.values[device.type.index]
        ..remoteId = device.id.toString();
      prefs.setString('device', proto.writeToJson());
    } else {
      prefs.setString('device', null);
    }
  }
```

In my code, I have a listener that runs `device.connect()` when the device is set.  I run _loadState() when my application starts, and it appears to be working.  I think the API could be improved by making the serialization more exposed, but the functionality is there.

FWeissenb, 2019-07-05:
The problem is not android. It's an iOS security issue that we can't access peripherals without scanning. (btw it's the same with application document path, its generated new so we cant save an absolute path to an image for example).

@pauldemarco Implementing this (like @fvisticot  suggested )in the library would be nice but I think this should be done by an iOS developer. I don't have any experience with iOS, I only use a mac to create my iOS builds :>

My use case (I think the same like @fvisticot and @assumptionsoup):
1. Device search in our app.
2. Serialize device parameter to database/file/wahtever
3. Close app
4. Restart app
5 Deserialize device parameter
6. Reconnect device, discover services and register notifications, read characteristics, ...

So everything works fine without step 6 :)

RFlame, 2019-07-14:
Is there a way to solve this problem??？

FWeissenb, 2019-08-13:
@pauldemarco any news about this?

yuriboeira11tx, 2021-04-20:
@pauldemarco any news about this?