pauldemarco/flutter_blue #274: how does _setNotification work?

DWShuo, 2019-06-16:
Im a little confused on how _setNotification (from the example code) works, i am pretty new to flutter/dart so these might seem like a really dumb questions.
- where is this "d" variable coming from? i don't see it being declared anywhere in scope
- what does the function setState do? also what is this syntax its using? ( ( ) { }) is this some sort of anonymous function type of deal?

```
_setNotification(BluetoothCharacteristic c) async {

    if (c.isNotifying) {

      await device.setNotifyValue(c, false);

      // Cancel subscription

      valueChangedSubscriptions[c.uuid]?.cancel();

      valueChangedSubscriptions.remove(c.uuid);
} else {
      await device.setNotifyValue(c, true);

      // ignore: cancel_subscriptions

      final sub = device.onValueChanged(c).listen((d) {

        setState(() {

          print(utf8.decode(d));

        });

      });

      // Add to map

      valueChangedSubscriptions[c.uuid] = sub;

    }
    setState(() {});
  }
```

khainke, 2019-06-16:
d is the variable defined by the anonymous closure for the listen callback. Basically listen gives you a streamsubscription and you supply it with a callbackhandler. For your code snippet: Whenever theres a value change on characteristic c, the callback will get the data in d as List<int> (this is omitted, dart can deduce it automatically). setState is called in stateful widgets to retrigger the build method.

See in particular this:
https://api.dartlang.org/stable/2.4.0/dart-async/StreamSubscription-class.html
https://pub.dev/documentation/flutter_blue/latest/flutter_blue/BluetoothDevice/onValueChanged.html

But you could also read up on Function, Future, Stream, StatefulWidget.

DWShuo, 2019-06-16:
oh nice thanks, I didn't know there were documentations for flutter blue

closed 2019-06-16 by DWShuo