pauldemarco/flutter_blue #276: discoverServices() fails on iOS

erichiggins, 2019-06-18:
## My code:

```
...
if (s == BluetoothDeviceState.connected) {
    try {
          store.state.deviceState.device.discoverServices().then((services) {
            BluetoothService service = services
              .where((s) => s.uuid.toString() == SERVICE_UUID).single;
            store.dispatch(SetService(service));
            _readCharacteristics(service?.characteristics, 0);
          }).catchError((err) { print(err); });
        } catch(err) {
          print(err);
        }
} else {
    disconnect();
}
...
```

## Error
```
2019-06-18 10:05:52.210212-0700 Runner[598:180453] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[0]'
```

erichiggins, 2019-06-30:
@pauldemarco Would it be possible to get some advice on this issue, please? Is iOS support broken w/ the latest version of flutter_blue?

LiangLuDev, 2019-11-07:
I had this same issue，Is there any solution?