pauldemarco/flutter_blue #279: Send String value in writeCharacteristic

IAmSarthakVerma, 2019-06-19:
Currently we can send List of Integers while writing a characteristic. I wanted to know whether we can send a String value instead of Integers.

IAmSarthakVerma, 2019-06-19:
> Currently we can send List of Integers while writing a characteristic. I wanted to know whether we can send a String value instead of Integers.

So currently I have converted the string to UTF8 and then sending that to the characteristic write function. But this increases the overhead on the other side too that the utf8 values have to be converted and a string is made out of it.

pauldemarco, 2019-06-19:
@IAmSarthakVerma This is a question better suited for a QA site like [stackoverflow](https://stackoverflow.com/), where it can be indexed and available to others.

If you feel this is a bug with the plugin, please re-open the issue explaining your situation.

closed 2019-06-19 by pauldemarco

IAmSarthakVerma, 2019-06-19:
@pauldemarco I was just of the opinion that there should be an option in the plugin itself to send different DataTypes instead of just bytes.

pauldemarco, 2019-06-19:
@IAmSarthakVerma I see, so that comes down to whether the transformation should happen externally to the library or within the library.

In my opinion, to keep the library thin at this time, I would prefer to leave it up to the end-user to take whatever type of data they have, and get it into bytes using any method they so choose.

Otherwise, it will be hard to draw the line at which types to give first-class write support, and in what format they should be converted.

I am obviously open to any thoughts on this, and perhaps we should open up a more general issue to investigate this feature.