pauldemarco/flutter_blue #281: Streams and Sinks only API

pauldemarco, 2019-06-19:
Consider an API where only Streams and Sinks are exposed:

1. If you'd like to write to a characteristic, you emit an item to that characteristic value's Sink
2. If you'd like to read a characteristic, you subscribe to the characteristic value's Stream

This would remove the explicit calls `read` and `write`. For instance, if you'd like to only read a value once, you can use a one-shot stream call like `characteristic.value.first`.

A separate branch should be created to investigate this API.

pauldemarco, 2019-06-19:
This has the added benefit of making stream transformations that much easier.

For instance, one could simply pipe the value stream's List<int> to a String converter to handle conversion automatically. Vice versa, the value's sink could be wrapped with a String converter to automatically take String input from the user and convert to List<int>. (https://github.com/pauldemarco/flutter_blue/issues/279)

bpillon, 2019-06-19:
@pauldemarco May be some values can be ValueListenable instead of Stream ? Even if "only streams api" is attractive, for some simple values like BluetoothState or BluetoothDeviceState, ValueListenable may be lighter for the same result.

shinayser, 2019-08-01:
> @pauldemarco May some values can be ValueListenable instead of Stream ? Even if "only streams api" is attractive, for some simple values like BluetoothState or BluetoothDeviceState, ValueListenable may be lighter for the same result.

Streams can be wrapped into a Observable and thus, receive all the benefitx of rxdart transformation methods. ValueListenable, don't.