pauldemarco/flutter_blue #282: [Version 0.6.0+1] Can't build APK on mac

happy49sky, 2019-06-20:
$ flutter build apk
Initializing gradle...                                              1.8s
Resolving dependencies...                                          15.3s
Running Gradle task 'assembleRelease'...                                                                                                                                                                                                                                 
FAILURE: Build failed with an exception.                                                                           
                                                                                                                   
* What went wrong:                                                                                                 
Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_protoc'.                         
> Could not download protoc-osx-x86_64.exe (com.google.protobuf:protoc:3.7.0)                                      
   > Could not get resource 'https://jcenter.bintray.com/com/google/protobuf/protoc/3.7.0/protoc-3.7.0-osx-x86_64.exe'.
      > Could not GET 'https://jcenter.bintray.com/com/google/protobuf/protoc/3.7.0/protoc-3.7.0-osx-x86_64.exe'.  
         > Connect to d29vzk4ow07wi7.cloudfront.net:443 [d29vzk4ow07wi7.cloudfront.net/99.84.231.221, d29vzk4ow07wi7.cloudfront.net/99.84.231.76, d29vzk4ow07wi7.cloudfront.net/99.84.231.174, d29vzk4ow07wi7.cloudfront.net/99.84.231.93] failed: Read timed out
                                                                                                                   
* Try:                                                                                                             
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
                                                                                                                   
* Get more help at https://help.gradle.org                                                                         
                                                                                                                   
BUILD FAILED in 2m 32s                                                                                             
Running Gradle task 'assembleRelease'...                                                                           
Running Gradle task 'assembleRelease'... Done                     153.1s (!)
Gradle task assembleRelease failed with exit code 1

pauldemarco, 2019-06-20:
Looks like both of those links are active downloads. Are you sure your connection isn’t being blocked for some reason?

happy49sky, 2019-06-20:
Does "protoc-3.7.0-osx-x86_64.exe" can run on MacOS ? 

I can download 'https://jcenter.bintray.com/com/google/protobuf/protoc/3.7.0/protoc-3.7.0-osx-x86_64.exe' by wget.

$ wget https://jcenter.bintray.com/com/google/protobuf/protoc/3.7.0/protoc-3.7.0-osx-x86_64.exe
--2019-06-20 16:47:54--  https://jcenter.bintray.com/com/google/protobuf/protoc/3.7.0/protoc-3.7.0-osx-x86_64.exe
Resolving localhost (localhost)... 127.0.0.1
Connecting to localhost (localhost)|127.0.0.1|:8123... connected.
Proxy request sent, awaiting response... 302
Location: https://d29vzk4ow07wi7.cloudfront.net/a2d703365c4e4bfec62204108eaa6a06ca8307ca93cae4d0713a4eafcf23cddb?response-content-disposition=attachment%3Bfilename%3D%22protoc-3.7.0-osx-x86_64.exe%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvYTJkNzAzMzY1YzRlNGJmZWM2MjIwNDEwOGVhYTZhMDZjYTgzMDdjYTkzY2FlNGQwNzEzYTRlYWZjZjIzY2RkYj9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMnByb3RvYy0zLjcuMC1vc3gteDg2XzY0LmV4ZSUyMiIsIkNvbmRpdGlvbiI6eyJEYXRlTGVzc1RoYW4iOnsiQVdTOkVwb2NoVGltZSI6MTU2MTAyMTE5NX0sIklwQWRkcmVzcyI6eyJBV1M6U291cmNlSXAiOiIwLjAuMC4wLzAifX19XX0_&Signature=Qztg--JHF055dBjbYh1yvInFBkCM762kUttlcX~hm2tiyk1DISoi~8uG4jg3lqJySeoEysOkeHJWVFLN0G9qezk9WsMsa8euFLXCO7aDFWvWIaRTPBRxSQu0FE7eUUSMjH~qcHiEWzBMy05Lw4DuheLOBe4-lFKlMbp9KfIXIiAacX9pJJN7l1wI~BjEA1SrsKq~q4bw~04sRoEYy0F-EkVQwgekLCb6DjkN3-YK9yCOPEPurPlpFuyXD-EQctNiyWo-zJDryZpjwKSCinf8Bjmw~fxwzTR9qNZqkd7Bh1kesd0xS2dCRV4nKS~H~aVYt0jxin-GmGTaMFDRBCdKFA__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA [following]
--2019-06-20 16:47:55--  https://d29vzk4ow07wi7.cloudfront.net/a2d703365c4e4bfec62204108eaa6a06ca8307ca93cae4d0713a4eafcf23cddb?response-content-disposition=attachment%3Bfilename%3D%22protoc-3.7.0-osx-x86_64.exe%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvYTJkNzAzMzY1YzRlNGJmZWM2MjIwNDEwOGVhYTZhMDZjYTgzMDdjYTkzY2FlNGQwNzEzYTRlYWZjZjIzY2RkYj9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMnByb3RvYy0zLjcuMC1vc3gteDg2XzY0LmV4ZSUyMiIsIkNvbmRpdGlvbiI6eyJEYXRlTGVzc1RoYW4iOnsiQVdTOkVwb2NoVGltZSI6MTU2MTAyMTE5NX0sIklwQWRkcmVzcyI6eyJBV1M6U291cmNlSXAiOiIwLjAuMC4wLzAifX19XX0_&Signature=Qztg--JHF055dBjbYh1yvInFBkCM762kUttlcX~hm2tiyk1DISoi~8uG4jg3lqJySeoEysOkeHJWVFLN0G9qezk9WsMsa8euFLXCO7aDFWvWIaRTPBRxSQu0FE7eUUSMjH~qcHiEWzBMy05Lw4DuheLOBe4-lFKlMbp9KfIXIiAacX9pJJN7l1wI~BjEA1SrsKq~q4bw~04sRoEYy0F-EkVQwgekLCb6DjkN3-YK9yCOPEPurPlpFuyXD-EQctNiyWo-zJDryZpjwKSCinf8Bjmw~fxwzTR9qNZqkd7Bh1kesd0xS2dCRV4nKS~H~aVYt0jxin-GmGTaMFDRBCdKFA__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA
Connecting to localhost (localhost)|127.0.0.1|:8123... connected.
Proxy request sent, awaiting response... 200 OK
Length: 10194380 (9.7M) [application/content-stream]
Saving to: ‘protoc-3.7.0-osx-x86_64.exe.1’

protoc-3.7.0-osx-x86_64.exe.1        12%[=======>                                                            ]   1.23M   555KB/s

happy49sky, 2019-06-20:
I can build now. Thanks. It's network issue.

closed 2019-06-20 by happy49sky