pauldemarco/flutter_blue #283: [0.6.0+1] Disconnect device

bpillon, 2019-06-20:
Hello Paul,
Running the sample. Trying to disconnect from a BluetoothDevice, it seems it's not working anymore while it worked well in 0.5.0.

bretie, 2019-06-21:
I'm having the same issue. I've tried canceling the device stream subscription and setting device connection to null. But I still get a issue with the connection not canceling.

tortechnocom, 2019-07-05:
I confirmed it works fine on latest version 0.6.0+1

GauravPatni, 2019-07-18:
I'm having the same issue even with 0.6.0+1. 
on example app, on-screen it shows connect & disconnect (after triggering action menu button ) but from the device side, it always remains in a connected state. 
The only way to disconnect is to disable "Bluetooth".

RFlame, 2019-07-22:
I'm also having the same issue even with 0.6.0+1.